bdring/Grbl_Esp32 #269: ESP resets when trying to connect to the ESP web server from pc

kjjordans, 2019-10-27:
Please answer the following questions.

What version of the firmware are you using?
_Master from 4 days ago_
Is the problem repeatable?
_yes_
Under what conditions does the bug occur?
_when i try to connect to the web interface the esp resets. 
I have tried to recompile without any edits to the config and i am still getting the same error.
I have disconnected all hardware and tried the CPU_MAP_TEST_DRIVE config and i am still getting the same error._ 

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[MSG:Axis count 3]
[MSG:RMT Steps]

[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Using cpu_map...CPU_MAP_PEN_LASER]

Grbl 1.1f ['$' for help]
E (6120) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (6120) task_wdt:  - IDLE0 (CPU 0)
E (6120) task_wdt: Tasks currently running:
E (6120) task_wdt: CPU 0: serialCheckTask
E (6120) task_wdt: CPU 1: loopTask
E (6120) task_wdt: Aborting.
abort() was called at PC 0x400fb8d8 on core 0

Backtrace: 0x40092980:0x3ffbe320 0x40092bb1:0x3ffbe340 0x400fb8d8:0x3ffbe360 0x40084da1:0x3ffbe380 0x4000bfed:0x3ffd29a0 0x40090235:0x3ffd29b0 0x4008eaad:0x3ffd29d0 0x40085c8d:0x3ffd2a10 0x40085d43:0x3ffd2a40 0x40154e19:0x3ffd2a60 0x401f9901:0x3ffd2ab0 0x400ea90b:0x3ffd2ad0 0x400ea7c2:0x3ffd2b10 0x400e32ce:0x3ffd2b30 0x400e3471:0x3ffd2b60 0x4021ade1:0x3ffd2c20 0x400f147a:0x3ffd2c40 0x400f1521:0x3ffd2c60 0x400f158e:0x3ffd2c90 0x400f16f7:0x3ffd2ce0 0x400e20ad:0x3ffd2d30 0x400e97aa:0x3ffd2d70 0x400e93dc:0x3ffd2d90 0x400dee31:0x3ffd2db0 0x4008f0c9:0x3ffd2dd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[MSG:Axis count 3]
[MSG:RMT Steps]

[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Using cpu_map...CPU_MAP_PEN_LASER]

luc-github, 2019-10-27:
did you checked the solution in FAQ ?

kjjordans, 2019-11-02:
Hi,
I have just gone through the FAQ as suggested. I only needed to do the first step 
ie. enter this into the serial monitor
[ESP710]FORMAT

Thanks for your help.

closed 2019-11-02 by kjjordans