bdring/Grbl_Esp32 #270: Modified CPU map not working

mac7988, 2019-10-27:
I made changes to the CPUMAP and nothing else. When I upload it nothing seems to work. For now I am only trying to get X axis to work so don't pay attention to the rest of the pin maps.

Do I need to adjust something else before compiling and uploading?
`
  
/*
  cpu_map.h - Header for system level commands and real-time processes
  Part of Grbl
  Copyright (c) 2014-2016 Sungeun K. Jeon for Gnea Research LLC
  
  2018 -  Bart Dring This file was modified for use on the ESP32
          CPU. Do not use this with Grbl for atMega328P
  
  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.
  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.
  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifndef cpu_map_h
//#define cpu_map_h 

  /*
  Not all pins can can work for all functions.
  Check features like pull-ups, pwm, etc before
  re-assigning numbers
  
  (gpio34-39) are inputs only and don't have software pullup/down functions
  You MUST use external pull-ups or noise WILL cause problems.
  
  Unlike the AVR version certain pins are not forced into the same port. 
  Therefore, bit masks are not use the same way and typically should not be 
  changed. They are just preserved right now to make it easy to stay in sync
  with AVR grbl
  
  */
  
//Set your pin definition 
//let -1 to use default board pin
#define GRBL_SPI_SS -1
#define GRBL_SPI_MOSI -1
#define GRBL_SPI_MISO -1
#define GRBL_SPI_SCK -1
//Set your frequency 
#define GRBL_SPI_FREQ  4000000

  #define LIMIT_MASK 0  // no limit pins    
#ifdef CPU_MAP_MPCNC  // all versions...select below
  // This is the CPU Map for the Buildlog.net MPCNC controller
  
  // uncomment ONE of the following versions
  //#define V1P1
  #define V1P2  // works for V1.2.1 as well
    
  #ifdef V1P1
    #define CPU_MAP_NAME "CPU_MAP_MPCNC_V1P1"
  #else // V1P2
    #define CPU_MAP_NAME "CPU_MAP_MPCNC_V1P2"
  #endif  
    
  // switch to the correct default settings
  #ifdef DEFAULTS_GENERIC
    #undef DEFAULTS_GENERIC
  #endif
  #define DEFAULTS_MPCNC
  
  #define USE_GANGED_AXES // allow two motors on an axis 
  
  #define X_STEP_PIN      GPIO_NUM_4
  #define X_STEP_B_PIN    GPIO_NUM_22  // ganged motor
  #define X_AXIS_SQUARING
  
  #define Y_STEP_PIN      GPIO_NUM_14
  #define Y_STEP_B_PIN    GPIO_NUM_21  // ganged motor
  #define Y_AXIS_SQUARING
  
  #define Z_STEP_PIN      GPIO_NUM_27
  
  #define X_DIRECTION_PIN   GPIO_NUM_15
  #define Y_DIRECTION_PIN   GPIO_NUM_25  
  #define Z_DIRECTION_PIN   GPIO_NUM_33 
  
  // OK to comment out to use pin for other features
  #define STEPPERS_DISABLE_PIN GPIO_NUM_13        
        
  // Note: if you use PWM rather than relay, you could map GPIO_NUM_2 to mist or flood 
  //#define USE_SPINDLE_RELAY
  
  #ifdef USE_SPINDLE_RELAY
    #ifdef V1P1
      #define SPINDLE_PWM_PIN    GPIO_NUM_17
    #else // V1p2
      #define SPINDLE_PWM_PIN    GPIO_NUM_2
    #endif      
  #else
    #define SPINDLE_PWM_PIN    GPIO_NUM_16
    #define SPINDLE_ENABLE_PIN  GPIO_NUM_32
  #endif
  
  #define SPINDLE_PWM_CHANNEL 0
  // PWM Generator is based on 80,000,000 Hz counter
  // Therefor the freq determines the resolution
  // 80,000,000 / freq = max resolution
  // For 5000 that is 80,000,000 / 5000 = 16000 
  // round down to nearest bit count for SPINDLE_PWM_MAX_VALUE = 13bits (8192)
  #define SPINDLE_PWM_BASE_FREQ 5000 // Hz
  #define SPINDLE_PWM_BIT_PRECISION 8   // be sure to match this with SPINDLE_PWM_MAX_VALUE
  #define SPINDLE_PWM_OFF_VALUE     0
  #define SPINDLE_PWM_MAX_VALUE     255 // (2^SPINDLE_PWM_BIT_PRECISION)
  
  #ifndef SPINDLE_PWM_MIN_VALUE
      #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
  #endif
  
  // Note: Only uncomment this if USE_SPINDLE_RELAY is commented out.
  // Relay can be used for spindle or either coolant
  //#define COOLANT_FLOOD_PIN   GPIO_NUM_2
  //#define COOLANT_MIST_PIN    GPIO_NUM_2
  
  #ifdef V1P1 //v1p1
    #define X_LIMIT_PIN       GPIO_NUM_2
  #else
    #define X_LIMIT_PIN       GPIO_NUM_17
  #endif  
  
  //#define Y_LIMIT_PIN       GPIO_NUM_4  
  //#define Z_LIMIT_PIN       GPIO_NUM_15   
  #define LIMIT_MASK        B111
  
  #ifdef V1P2
    #ifndef ENABLE_SOFTWARE_DEBOUNCE   // V1P2 does not have R/C filters
      #define ENABLE_SOFTWARE_DEBOUNCE
    #endif
  #endif
  
  #define PROBE_PIN         GPIO_NUM_35  
  
  // The default value in config.h is wrong for this controller
  #ifdef INVERT_CONTROL_PIN_MASK
    #undef INVERT_CONTROL_PIN_MASK      
  #endif
  
  #define INVERT_CONTROL_PIN_MASK   B1110
  
  // Note: default is #define IGNORE_CONTROL_PINS in config.h
  // uncomment to these lines to use them   
  #ifdef IGNORE_CONTROL_PINS
    #undef IGNORE_CONTROL_PINS
  #endif
  
  #define CONTROL_RESET_PIN         GPIO_NUM_34  // needs external pullup
  #define CONTROL_FEED_HOLD_PIN     GPIO_NUM_36  // needs external pullup 
  #define CONTROL_CYCLE_START_PIN   GPIO_NUM_39  // needs external pullup       
    
#endif

   

  // ================= common to all machines ================================
  
  // These are some ESP32 CPU Settings that the program needs, but are generally not changed
    #define F_TIMERS  80000000    // a reference to the speed of ESP32 timers
    #define F_STEPPER_TIMER 20000000  // frequency of step pulse timer
    #define STEPPER_OFF_TIMER_PRESCALE 8 // gives a frequency of 10MHz
    #define STEPPER_OFF_PERIOD_uSEC  3  // each tick is
    
    #define STEP_PULSE_MIN 2   // uSeconds
    #define STEP_PULSE_MAX 10  // uSeconds
    
    // =============== Don't change or comment these out ======================
    // They are for legacy purposes and will not affect your I/O 
    
    #define X_STEP_BIT    0  // don't change
    #define Y_STEP_BIT    1  // don't change
    #define Z_STEP_BIT    2  // don't change    
    #define A_STEP_BIT    3  // don't change
    #define B_STEP_BIT    4  // don't change
    #define C_STEP_BIT    5  // don't change
    #define STEP_MASK       B111111 // don't change
    
    #define X_DIRECTION_BIT   0 // don't change
    #define Y_DIRECTION_BIT   1  // don't change
    #define Z_DIRECTION_BIT   2  // don't change
    #define A_DIRECTION_BIT   3 // don't change
    #define B_DIRECTION_BIT   4  // don't change
    #define C_DIRECTION_BIT   5  // don't change

    #define X_LIMIT_BIT       0  // don't change
    #define Y_LIMIT_BIT       1  // don't change
    #define Z_LIMIT_BIT       2  // don't change
    #define A_LIMIT_BIT       3  // don't change
    #define B_LIMIT_BIT       4  // don't change
    #define C_LIMIT_BIT       5  // don't change    
    
    #define PROBE_MASK        1 // don't change   
    
    #define CONTROL_MASK              B1111   // don't change
    
    // =======================================================================
    
    
#endif
`

bdring, 2019-10-27:
I don't understand. 

 1. Did you modify the MPCNC cpu map?
 2. What hardware are you running on?
 3. What did you modify?
 4. Did you change anything in config.h? 
 5. Can you paste the startup text you get in the serial monitor?

mac7988, 2019-10-27:
Did you modify the MPCNC cpu map?
Yes, I have a different pin layout.
What hardware are you running on?
It's a board I soldered to test. Using esp32 devkit c

What did you modify?
Just the CPU map
Did you change anything in config.h?
No
Can you paste the startup text you get in the serial monitor

mac7988, 2019-10-27:
I figured it out. all I had to do is read what you wrote on the WIKI.
I am so impressed by how well you wrote this. Even for a dummy like me.

What I did is change the config to to use the correct map. In this case the mpcnc map that I modified.

closed 2019-10-27 by mac7988