bdring/Grbl_Esp32 #271: SD Card Reader Suggestions

jaysettle, 2019-10-28:
I'm interested in running jobs untethered on the ESP32 I'll need an SD card reader. 

I'm also not using the PCB you created, what SD card reader would you suggest if any? Thanks.

[image]

bdring, 2019-10-28:
You mean micro SD card socket?

On the BOM J20

https://github.com/bdring/Grbl_ESP32_Development_Controller/blob/master/docs/V3p5/Bom_3p5%20-%20Bom_3p1.csv

ithinkido, 2019-10-28:
Any chance of being able to use something like this...
https://m.alibaba.com/product/60672650653/Micro-SD-Card-Module-Mini-TF.html 
?

bdring, 2019-10-28:
Yes, I have used that. Look for the schematic in that repo on how to hook it up.

closed 2019-11-04 by bdring