bdring/Grbl_Esp32 #272: Let's Pause For A Second

jaysettle, 2019-10-29:
Barton Dring I was looking at your youtube channel today and I was very impressed. Not even by the projects but the range of talent here. 
-3D designing and printing
-Finished project videos with an actual intellectual describing the project without bragging
-Circuit design and implementation with webpage to sell your boards
-Software knowledge to port GRBL to ESP32 (big wow here)
-All of this is presented and documented so well in a wiki (you need to add a pin mapping damnit)
-All my dumb questions are answered the day I post the in the forums
-And from your youtube videos you seem to have an artistic side too

Just want to say I see you and thanks. I feel like you're in the zone I hope these people are donating enough to make it worth it(probably not).

-Jay

f4grx, 2019-10-29:
this project is so M8 !

shreeramlive, 2019-12-17:
> [G4 P1.0](http://www.linuxcnc.org/docs/html/gcode/g-code.html#gcode:g4)
> 
> Thanks

I took me fair amount of time to understand this😆 Nice joke BTW

closed 2020-01-15 by bdring