bdring/Grbl_Esp32 #273: Laser PWM

AndyLuz, 2019-10-30:
Hello
Would it be possible to implement PWM frequency setting for lasers like grbl-LPC
 $33? It would be very useful to be able to change the frequency without recompiling the firmware.

Also thank you for great job..

bdring, 2019-10-30:
I will look into it.  Can you give the variable types for all those values.

Float, int8, int16, etc.

bdring, 2019-10-31:
Found the information myself. Your link got me close to it....all floats

New configuration settings

$33 is PWM frequency in Hz
$34 is PWM off value in %
$35 is PWM min value in %
$36 is PWM max value in %
$140, $141, $142 are X, Y, Z current (amps)

```C++
typedef struct {
  // Axis settings
  float steps_per_mm[N_AXIS];
  float max_rate[N_AXIS];
  float acceleration[N_AXIS];
  float max_travel[N_AXIS];
  float current[N_AXIS];

  // Remaining Grbl settings
  uint8_t pulse_microseconds;
  uint8_t step_invert_mask;
  uint8_t dir_invert_mask;
  uint8_t stepper_idle_lock_time; // If max value 255, steppers do not disable.
  uint8_t status_report_mask; // Mask to indicate desired report data.
  float junction_deviation;
  float arc_tolerance;

  float spindle_pwm_freq;      // Hz
  float spindle_pwm_off_value; // Percent
  float spindle_pwm_min_value; // Percent
  float spindle_pwm_max_value; // Percent
  float rpm_max;
  float rpm_min;

  uint8_t flags;  // Contains default boolean settings

  uint8_t homing_dir_mask;
  float homing_feed_rate;
  float homing_seek_rate;
  uint16_t homing_debounce_delay;
  float homing_pulloff;
} settings_t;
```

bdring, 2019-10-31:
I think I will have a way to turn off display of those settings. like...

```C++
  #define SHOW_EXTENDED_SETTING
```

I am afraid it will break some senders. Non robust GCode senders generate a lot of support questions.

The default will be off. You could send $33=5000 to set PWM frequency in Hz in any mode, but $$ will only show the original settings.

bdring, 2019-11-03:
This is complete and on the master branch

closed 2019-11-04 by bdring

AndyLuz, 2019-11-12:
Wow
Sorry for late reply. My computer crashed... That was fast. I will test this on my CO2 laser as soon as possible.