bdring/Grbl_Esp32 #275: How to get external  buttons working

mac7988, 2019-10-31:
I have external buttons hooked up and mapped to the pins or Cycle start, Feed hold and Reset.

Seem to not work. I am sure doing something wrong in the config. All I did is assign my pins and cpu map. Do I need to do anything else? Below is my  wiring diagram for 1 button.
[image]

ggallant571, 2019-11-01:
> image

Seems overly complicated. Why not use the external switch ground the 
input to U1.

mac7988, 2019-11-01:
Yes you are right. It's not fun to be simple. haha

mac7988, 2019-11-01:
Has anyone tested external buttons? I  tried to simply connect button between gpio and GND and still nothing(WTF).

bdring, 2019-11-01:
I use them all the time. Are they reading in the ? Command?

mac7988, 2019-11-01:
No I get nothing. I use pin 33 for cycle start, 34 for Feed hold and 35 for reset. Like I said I even used a push button hooked up to 33 and gnd and still nothing. Kind of confused.

bdring, 2019-11-01:
Do you have pullups?

bdring, 2019-11-01:
Check config.h for the IGNORE

mac7988, 2019-11-01:
I check what you said about the commands in UGS and it didn't show anything . I did the same in arduino IDE serial and it should a pin x pressed.

[image]

mac7988, 2019-11-01:
OK the ignore was active disabled it and now doesn't matter what I press all buttons are reset via control pin. 

[image]

mac7988, 2019-11-03:
I tried multiple pins and still getting reset issues. I am officially lost. I even tried different esp23 boards with the same result. I also tried different computers, different cables and even OTG.

Below test was done with nothing connected esp23 board only and a jumper cable attached to pin 36. I can't imagine others don't have this probelm. 
[image]

bdring, 2019-11-03:
Pull up resistors?

mac7988, 2019-11-03:
No pull up resistors

mac7988, 2019-11-03:
all I have is a esp32 and a dupont jumper hooked up to pin 36 in you touch the cable it freaks out(this has happen to multiple esp32s

bdring, 2019-11-03:
https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins

Pins 34 thru 39 need pullups

mac7988, 2019-11-03:
Tried that, I even tried pin 22 as cycle start also on multiple esp32. Do you think I have an issue with libraries being old or corrupted?

bdring, 2019-11-03:
Your symptoms are exactly like a floating input.  I think your problem is electrical

mac7988, 2019-11-03:
The weird thing is I don't have anything hooked up.

[image]

bdring, 2019-11-03:
I am testing now.  Hang on. I might see a problem.

bdring, 2019-11-03:
I think I found a problem...

For some reason the internal pullups were [turned off recently](https://github.com/bdring/Grbl_Esp32/blame/6ebf56f0ee91be4600f6f8e38e125fb24036b916/Grbl_Esp32/system.cpp#L37).

In system.cpp change this section in system.ini()
pinMode(....INPUT_PULLUP);

```C++
#ifdef CONTROL_SAFETY_DOOR_PIN
	pinMode(CONTROL_SAFETY_DOOR_PIN, INPUT_PULLUP);
	attachInterrupt(digitalPinToInterrupt(CONTROL_SAFETY_DOOR_PIN), isr_control_inputs, CHANGE);
#endif
#ifdef CONTROL_RESET_PIN
	pinMode(CONTROL_RESET_PIN, INPUT_PULLUP);
	attachInterrupt(digitalPinToInterrupt(CONTROL_RESET_PIN), isr_control_inputs, CHANGE);
#endif
#ifdef CONTROL_FEED_HOLD_PIN
	pinMode(CONTROL_FEED_HOLD_PIN, INPUT_PULLUP);
	attachInterrupt(digitalPinToInterrupt(CONTROL_FEED_HOLD_PIN), isr_control_inputs, CHANGE);
#endif
#ifdef CONTROL_CYCLE_START_PIN
	pinMode(CONTROL_CYCLE_START_PIN, INPUT_PULLUP);
	attachInterrupt(digitalPinToInterrupt(CONTROL_CYCLE_START_PIN), isr_control_inputs, CHANGE);
#endif

```

In config.h enable turn on control switch debouncing.
```C++
#define ENABLE_CONTROL_SW_DEBOUNCE
```

Modify in cpu_map.h the first map to add a reset button. This is the simplest cpu_map. I use 21 here, but try your pin too. Use an external pullup if in the range 34 to 39.
**Be sure to use this map in config.h**

```C++
#ifdef CPU_MAP_TEST_DRIVE
	
	#define CPU_MAP_NAME "CPU_MAP_DEFAULT - Demo Only No I/O!"	
	
	// the following items currently need to be defined, but no i/o needs to be mapped
	// fixing soon
	
	#define CONTROL_RESET_PIN         GPIO_NUM_21 
	
	#define LIMIT_MASK 0  // no limit pins		
#endif
```

mac7988, 2019-11-03:
I tested the new code and found that it is not crashing but the only thing I am getting now is below no matter which button I press. It is reseting only

[image]

mac7988, 2019-11-03:
I even triple check my wiring see below. I even bypassed before optocoupler and just used a jumper between gnd and pullup . No luck :( still reset via control pin.
[image]

[image]

bdring, 2019-11-03:
What cpu mask are you using? Please paste the cpu map here. Just your part, not the whole file.

mac7988, 2019-11-03:
I think I got it working turns our my pulls were not connected.

closed 2019-11-04 by mac7988