bdring/Grbl_Esp32 #276: Loop a gcode

mac7988, 2019-11-01:
loop a gcode?

I think for me it would really help if I can load a gcode and keep looping it. Let me know what everyone else thinks or is there already a way to do this.

bdring, 2019-11-01:
It is not something I am interested in.

In the future, please use the "Feature Request" template.

misan, 2019-11-02:
@mac7988 given that SD card size is pretty big compared to usual gcode file size, you can create a gcode file that contains as many loop iterations as you need and then use that file for your purpose. 
https://superuser.com/questions/338616/linux-command-to-concatenate-a-file-to-itself-n-times

ithinkido, 2019-11-02:
> loop a gcode?
> 
> I think for me it would really help if I can load a gcode and keep looping it. Let me know what everyone else thinks or is there already a way to do this.

Are you trying to do something like this ?https://youtu.be/4xNMYLPE_jM

mac7988, 2019-11-04:
> > loop a gcode?
> > I think for me it would really help if I can load a gcode and keep looping it. Let me know what everyone else thinks or is there already a way to do this.
> 
> @mac7988 Are you trying to do something like this ?https://youtu.be/4xNMYLPE_jM

Yes, I know grbl can't handle sub routine. I will most likely store it in the sd card and have a macro to cycle start the program using a second Arduino.

It's crude but let's see what happens.

bdring, 2019-11-04:
I have always wanted to play with those advanced looping features of gcode, but it is not practical for Grbl. 

Grbl does not store the lines. It does not have the RAM do that. You might be able to store an index on the SD card, but it would be a huge amount of work for a very small audience.

If you just want the file to repeat, you could probably hack that pretty easily, but I don't like to clutter up the main code for novelty things like that.

closed 2019-11-04 by bdring

ithinkido, 2019-11-05:
Bart, what do you think of this implementation. If I have understood it right, the g-code sender is responsible for processing the loop. It is a very different solution, but it is also possible idea.https://github.com/svenhb/GRBL-Plotter/wiki/subroutines

bdring, 2019-11-05:
That is a good solution.

QGB, 2023-06-06:
grbl 不支持的简单 WHILE-ENDWHILE 循环结构