bdring/Grbl_Esp32 #285: Can not control the motor

jasinghuang, 2019-11-08:
Hello, I just finished flash the board, and I can get into the web interface, but I can not control the XYZ motor, when I click the XYZ button in the web controller or GRBL controller(PC), it shows nothing. 

And it can not recognize sandisk TF 64g card.
[image]

bdring, 2019-11-08:
Please paste the text you get on your serial port at startup in a reply.

jasinghuang, 2019-11-08:
> Please paste the text you get on your serial port at startup in a reply.

[MSG:Client Started]
[MSG:Connecting TP Jasing 2.4]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connecting.]
[MSG:Disconnected]
[MSG:Connected with 192.168.6.107]
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Using cpu_map...CPU_MAP_DEFAULT - Demo Only No I/O!]

Grbl 1.1f ['$' for help]

mac7988, 2019-11-08:
You are using the wrong CPU map.  Change in the config.h
[image]

In your case to CPU_MAP_ESP32 ( I think)

jasinghuang, 2019-11-08:
> You are using the wrong CPU map. Change in the config.h
> [image]
> 
> In your case to CPU_MAP_ESP32 ( I think)

I have select the version 3.5 which I use in cpu_map.h

	// Select the version (uncomment one of them)
	#define CPU_MAP_V3p5 // version 3.5 and earlier
	//#define CPU_MAP_V4 // version 4 or higher (in developement)

bdring, 2019-11-08:
@jasinghuang The advice from @mac7988 is correct. 

Does it work now?

jasinghuang, 2019-11-09:
@bdring @mac7988 thank you both, I change to 
#define CPU_MAP_ESP32 // these are defined in cpu_map.h
It works now.

But it still can not detect TF card. 

[image]
[image]

bdring, 2019-11-09:
Did you try the formatting suggestions on the wiki?

https://github.com/bdring/Grbl_Esp32/wiki/Using-the-SD-Card

jasinghuang, 2019-11-10:
> Did you try the formatting suggestions on the wiki?
> 
> https://github.com/bdring/Grbl_Esp32/wiki/Using-the-SD-Card

I tried, it stil not working.

jasinghuang, 2019-11-10:
Finally, I try to format the card in FAT, and it works!!!

closed 2019-11-10 by bdring