bdring/Grbl_Esp32 #286: How to get Macro working

mac7988, 2019-11-08:
This is what I am done so far. 

1) I setup 2 files Simple.H  and Simple.ccp
  a) Simple.h contains 
           void user_defined_macro(uint8_t index);
* i think this is not needed.

 b) Simple.ccp contains

#ifdef CPU_MAP_TEST1

void user_defined_macro(uint8_t index)
{
switch (index) {
		#ifdef MACRO_BUTTON_0_PIN
		case CONTROL_PIN_INDEX_MACRO_0:
			inputBuffer.push("$H\r"); // home machine
		break;

}
}
#endif

2) In my CPU MAP I added 

#include "Simple.h"

#define MACRO_BUTTON_0_PIN		GPIO_NUM_22

I am not able to compile it. I am surely doing something wrong.

bdring, 2019-11-08:
Follow the atari_1020 as an example.

In cpu_map.h you should include your simple.h (keep it all lower case to match most other files) inside your cpu map.

In simple.h, place all your #define statement first. Then wrap all your function prototypes like this at the bottom.

```C++
#ifndef simple_h
	#define simple_h

#endif
 ```

in simple.cpp, include grbl.h at the top. Surround all your functions with the #ifdef of your cpu map. so it only uses them for your machine.

Paste your compile error if you still have trouble.

mac7988, 2019-11-08:
> Follow the atari_1020 as an example.
>
> In cpu_map.h you should include your simple.h (keep it all lower case to match most other files) inside your cpu map.
>
> In simple.h, place all your #define [...]

Thank you, I will try it when I get home.  Thank you.

mac7988, 2019-11-09:
Hi Bart,

Thank you for the info. Silly mistake but got it working following your suggestions.
I did notice if you press the macro to run a Gcode from the sd card it will run the code but from that point forward hitting a cycle start will also run that macro( even after a power cycle) not a problem for me, In fact it's good for a headless machine.

Also serial monitor shows invalid gcode error if you press feedhold. which is nothing in the code just weird. 

[image]

Did I mention your work here is amazing. I will do my best to support other users from my learning as well.

Cheers!

bdring, 2019-11-09:
Glad to hear it. The compiler is very picky and does not behave exactly the same as some other c compilers.

closing....

closed 2019-11-09 by bdring