bdring/Grbl_Esp32 #287: Pcb for 4axis (not spi) with external drivers

aldanstar, 2019-11-12:
Hello,
Is there pcb for esp32 to working with external drivers using X,Y,A,B pins added in the last firmware (x6 defined in cpumap)? If right, where we can buy it for own project. 
Thank you for your everything:)

shreeramlive, 2019-11-12:
You can use this [Grbl_ESP32 MPCNC CNC Controller Ver 1.2.1](https://www.tindie.com/products/33366583/grbl_esp32-mpcnc-cnc-controller-ver-121/) board.
But you have to connect the **Enable, Step, Dir** to your `external stepper drivers` directly ( some drivers may not be **3.3V** compliant, **5V** will damage ESP32 module). What drivers do you use?

[image]

This board was designed for **MPCNC** (only 3 axis). So you can configure an extra **Step** pin for the 4th axis **DIR signal**, This board already has **Step** signal pin for 5 steppers. You can modify the pin map to support the 4th Axis.

bdring, 2019-11-12:
You cannot use the MPCNC for 4 axis. The (2) X axes and the (2) Y axes share a direction pin. They cannot move in independent directions.

The cpu map you are talking about is for [this controller](https://github.com/bdring/Tap6x). I do not have any available for sale, but you could build your own from the source files.

karoria, 2019-11-12:
Hi Barton.. I am also interested in 4th Axis (A Axis). Can you share an example cpu map file? BTW, I am not using memory card pins if they can be used.. I am sure such example will be useful to other fellows also. I have noticed that all the examples with more than 3 axis are for hobby servo and not for stepper.

shreeramlive, 2019-11-12:
> You cannot use the MPCNC for 4 axis. The (2) X axes and the (2) Y axes share a direction pin. They cannot move in independent directions.
>
> The cpu map you are talking about is for [this [...]

@bdring Thanks for correction. I referred **MPCNC** board since @aldanstar mentioned to use **external stepper drivers**.

bdring, 2019-11-12:
I also have this design. I do not sell it, but the [documentation is here](https://github.com/bdring/4_Axis_SPI_CNC). I am considering selling some. 

It is for SPI drivers only and can handle the big TMC5160 drivers.

[image]