bdring/Grbl_Esp32 #288: Combining Feed and Rapid overrides

karoria, 2019-11-13:
Dear friends,
I am wondering if I can combine feed and rapid overrides. For example, if I send 0x92 (decrease feed by 10%), can it also reduce rapid by 10%? By the way, I am going to program a pro micro board for such shortcut keys. Should I program a pin to send 2 such ascii characters? Say 0x92 and 0x96? Or is there a coding way to just alter a source file and compile again?
Thanks in advance for your precious time.
Regards,
Ravi

mac7988, 2019-11-13:
Hi Ravi,
My thinking would be to send each command separate. Here is why, you want to get a return to see if feed rate changes or not and how much.

You can do this my monitoring verbose.

bdring, 2019-11-13:
I think sending the characters separately is the easiest way.

terjeio, 2019-11-13:
@karoria Be aware that rapid override is limited to two values, from [config.h](https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/config.h):

```
#define DEFAULT_RAPID_OVERRIDE  100 // 100%. Don't change this value.
#define RAPID_OVERRIDE_MEDIUM    50 // Percent of rapid (1-99). Usually 50%.
#define RAPID_OVERRIDE_LOW       25 // Percent of rapid (1-99). Usually 25%.
```

karoria, 2019-11-13:
Thanks for valuable suggestions. I can also try to program for a pin/button which sends both the commans one after another with a defined delay, say 50ms. I don't know how it would result. As @terjeio reminded, rapid overrides are defined in absolute manner, so it would be difficult for me to relate it with feed overrides. Any ideas or workaround to make rapid overrides incremental just like feed overrides?

bdring, 2019-11-13:
It might help if you give us a reason you are doing this. There are many ways to control speeds.

karoria, 2019-11-13:
I will use rotary encoder to make overrides user friendly just like we see in industrial machines. I am already using such VMC with Siemens system which has such combine override system which is more convenient to operate. Think about the machine is about to collide with job with good speed, in such case you have no time to guess whether it is feed or rapid move. You have to rotate single knob anticlockwise. Then you can have time for extra decision. Another knob is for spindle override. So, I want to put just 2 knobs as 3 knobs make it little more time consuming in extreme situations.

bdring, 2019-11-13:
OK, I understand.

BTW: The feed hold or the safety door features are designed for that.

karoria, 2019-11-14:
Right. It will surely prevent accidents. But after issuing feed hold, you need to set desired feed and rapid overrides and then issue cycle start for such cases is a long journey. I want to make it with a knob only. I am talking about first run of any new program. So operator can start with low feed and rapid and then gradually increase it and find the optimum one. That is why I take this more seriously.

derFrickler, 2019-11-14:
I know the feature from the ESTLCam machine control. Its quite nice to start the program with both feeds set 0 and then pitch it up to see if everything is ok.
But there i miss the feature to only slow down the G1/milling feedrate to just mill a bit slower but move as fast as possible.  Both options are useful for different stuff.

karoria, 2019-11-14:
@derFrickler 
I also thought that way. Now I am planning to combine both in rotary knob only (as mentioned earlier by issuing 2 ascii codes spaced at defined time interval). When I need separate overrides, I will use traditional way by clicking in the software. The only thing to workout is how to club that absolute override command of rapid with relative override command of feed.

karoria, 2019-11-16:
I am closing this now. In case you come across a way to make rapid overrides "incremental", kindly let me know anytime. Thanks again..

closed 2019-11-16 by karoria