bdring/Grbl_Esp32 #294: PWM question on MPCNC board

ericspooner, 2019-11-15:
Greetings,

I recently had bought an MPCNC board just before I saw a post about this board https://github.com/bdring/4_Axis_SPI_CNC. I'm using the board to control a diode laser and machine I built. Machine works great with an Uno/CNC Sheild, though I run into an issue of speed as while PWM for the laser to be on, I can get only about 1550 mm/m out of the atmega. I've been trying to get a machine that is able to do about 6000 mm/m while using the PWM and the ESP32 looked very promising. The other board above with the TMC drivers would be awesome as I would like to work on adding sensorless homing for my project. The less wires the better. 

That said, the laser is off ebay and has a 3Pin power/PWM connection. PWM is 5v, which I use the Z+ pin on the cnc shield for. This works exactly as expected. voltage ramps from 0-5v depending on the 0-255 you input to grbl. Now i'm still new and learning, but this seems like a combined behavior of TTL and PWM. I don't have an oscilloscope so I can't check the duty cycle on the pin output, but when I do the same with the ESP32 on the MPCNC board, I basically get a on/off. M3S1 gives me 100% power which is the same as M3S255 M5 turns it of, S0 doesn't turn it off. Am I missing a step? I'm really looking forward to the 4 axis board with SPI built in for TMC drivers. I will most likely be getting one to play when you you sell them to help support.

Thanks!

bdring, 2019-11-15:
The PWM issue could be several things. 

My first guess is you have it in or had it in `USE_SPINDLE_RELAY` mode. This changes the default max RPM to 1. This means all 'S' values of 1 or greater are full on. This is because, if you try to PWM a relay, you will break it quickly.

Look at the cpu map you used and make sure `#define USE_SPINDLE_RELAY`  is commented out. Also check your $30 setting. I would suggest $30=1000.

ericspooner, 2019-11-16:
Thank you very much for the response.
$30=255.000 is what I currently have on the config.

Under

```
#ifdef CPU_MAP_MPCNC  // all versions...select below
```

I have 

```
// Note: if you use PWM rather than relay, you could map GPIO_NUM_2 to mist or flood 
	//#define USE_SPINDLE_RELAY
```

Would adjusting any of these settings be able to get me to similar to the CNC shield on the arduino? 

```
	#define SPINDLE_PWM_CHANNEL 0
	// PWM Generator is based on 80,000,000 Hz counter
	// Therefor the freq determines the resolution
	// 80,000,000 / freq = max resolution
	// For 5000 that is 80,000,000 / 5000 = 16000 
	// round down to nearest bit count for SPINDLE_PWM_MAX_VALUE = 13bits (8192)
	#define SPINDLE_PWM_BASE_FREQ 5000 // Hz
	#define SPINDLE_PWM_BIT_PRECISION 8   // be sure to match this with SPINDLE_PWM_MAX_VALUE
	#define SPINDLE_PWM_OFF_VALUE     0
	#define SPINDLE_PWM_MAX_VALUE     255 // (2^SPINDLE_PWM_BIT_PRECISION)
	
	#ifndef SPINDLE_PWM_MIN_VALUE
			#define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
	#endif

```

Also I had found this [post./issue](https://github.com/bdring/Grbl_Esp32/issues/155) would something like [this amazon link](https://smile.amazon.com/NOYITO-Optocoupler-isolation-High-Power-Development/dp/B07R4VCCPL/ref=sr_1_8?crid=1KRACXZME1M04&keywords=optoisolator&qid=1573934032&sprefix=optoiso%2Caps%2C184&sr=8-8) be needed or should be used? What is interesting, when I put my multimeter on it, with it on the PWM pings, M3S0 gives ~1 volts same as M3S255. The only modifying I've done as I wanted to get it working before I started testing/playing to learn more was change the CPU MAP in config.h to CPU_MAP_MPCNC since that is the board. I noticed right away that I was getting 5v all the time, and in one of your screenshots the Intlk was jumpered, after doing that I only got voltage after running the m3 command, and m5 shuts it off correctly.

Again thank you for such an awesome project!!

bdring, 2019-11-18:
Are you about that $30 value? 

Send $$ to get all the $ settings and post them here.

ericspooner, 2019-11-18:
No, I'm not fixed to 255 at all, that was just my default on my Uno. I'm using lightburn for software so very easy to change that setting there as well. 

```
$$
$0=3
$1=250
$2=0
$3=0
$4=0
$5=1
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=3
$24=200.000
$25=2000.000
$26=250
$27=1.000
$30=255.000
$31=0.000
$32=1
$100=160.000
$101=160.000
$102=160.000
$110=6000.000
$111=6000.000
$112=3000.000
$120=2000.000
$121=2000.000
$122=1000.000
$130=375.000
$131=375.000
$132=300.000
```

bdring, 2019-11-18:
It hooked up an MPCNC controller and the current firmware appears to work fine. I have a few other questions and suggestions

1. What version of the MPCNC controller do you have. Look at the bottom of the board for the revision
2. Is the board connected to the laser while you are measuring?
3. Is the board externally powered during these measurements or just via USB?
4. Try sending $rst=$ to reset all settings.

ericspooner, 2019-11-18:
Sure thing,

1. Version 1.2.1
2. When showing ~5v connected, when showing ~1v no connected. Tried with laser connected when power goes to board/after board has booted as well.
3. Both. I have external power, I've use pronterface, as well as no USB and tried just the web interface. Results are the same in both tests. I have the external power there so I could make sure drivers work, and they work like champs.
4. I will give this a shot!

bdring, 2019-11-18:
It appears the laser may be pulling the signal down. The controller is 3.3V, so it uses has a 3.3V to 5V converter. That cannot drive something with a pull down, because it uses a pullup resistor for the 5V.

Can the laser module TTL accept a 3.3V signal and reach full power? We may be able to reroute a signal.

Are you planning to use both a a spindle and a laser. If so, is the spindle controlled by Grbl via relay or PWM?

ericspooner, 2019-11-18:
no, this board is only for the laser, no plans whatsoever to use a spindle on this board. This is the laser module, it's the 5500mw [Ebay link](https://www.ebay.com/itm/450nm-7W-Blue-Laser-Module-With-Heatsink-For-Laser-Cutter-Engraver-CNC-DIY-Laser/392206265262?ssPageName=STRK%3AMEBIDX%3AIT&var=661150079818&_trksid=p2060353.m1438.l2649) It doesn't have much information on what board they are using etc. From what I can tell, 3.3V doesn't seem to get it to full power, Had tried to use an BigTree Tech SKR 1.3 but the PWM pin is only 3.3v and wasn't able to get the same cut with it. I'm open to using other hardware to drive, but will be honest my knowledge in this is really lacking on using something like optoisolators pullup/down to get what we need. I'm still learning here, and sadly there isn't much I can find on it, and don't want to magic smoke my laser driver board as I'm working on christmas gifts with it :).

bdring, 2019-11-18:
I think an opto isolator could work. It would need to be a push, pull type so it can drive current on the 5V side. Most invert so it might be best to drive the LED "backwards", so low is high output.

Grbl_ESP32 can invert the PWM signal too.

[image]

ericspooner, 2019-11-18:
Thank you so much. I'm going to get some of these ordered and will try to set it up. If I have questions can I post them here? If you don't want the clutter I can keep trying to work it out :). Again awesome project, and thanks for sharing your knowledge!

bdring, 2019-11-18:
Here is fine, or Slack

ericspooner, 2019-11-19:
On a side note, I found out from the sell of my laser module that: Frequency: <15khz, PWM/TTL: 0-5V for the controller, so 3.3v won't make it full power. Does this help in anyway with our diagram?

ericspooner, 2019-11-21:
Got the parts in. I'm still a member of the fail army, wired up, and I still only get a 0 or 4.81v now, no inbetween. 

So decided to try normal again, however now I can't get even with re-uploading firmware it to not be inverted on the pin. I think something may be bonked with my ESP32. You wouldn't happen to have the link to which board builder  you used for your ESP32 dev board? I'm thinking that might be my issue all along is a bad board.

THanks!

bdring, 2019-11-21:
Try removing it from the controller PCB. Paste this CPU MAP into cpu_map.h. It is a super simple map that directs the spindle to the LED on some boards. You could change the I/O pin to others too.

```C++
#ifdef CPU_MAP_SPINDLE_TEST
	/*
	This is a simple cpu_map to test spindle function.
	
	Feel free to change the I/O pin, but GPIO_NUM_2 is often an
	LED on dev boards, so you can see it glow bright with high speeds
	and dim with low speeds
	
	send $rst=$ to reset all settings. RPM will be 0-1000
	
	*/
	#define CPU_MAP_NAME "Spindle Test Only"	
	
	#define SPINDLE_PWM_PIN    	GPIO_NUM_2
	#define SPINDLE_PWM_CHANNEL 0
	#define SPINDLE_PWM_BIT_PRECISION 8

	
	#define LIMIT_MASK 0  // no limit pins		
#endif
```

Change config.h to use this cpu map with 
```C++
#define CPU_MAP_SPINDLE_TEST
```

closed 2020-01-12 by bdring