bdring/Grbl_Esp32 #295: Getting only 2.7V output on Enable Pin

karoria, 2019-11-16:
Dear Friends,
I am facing problems with enable output. I am using STEPPERS_DISABLE_PIN GPIO_NUM_13. It goes high at 2.7V approx and low at 0. Due to this, my driver malfunctions (misses some pulses and enagages some pulses). Any clues why I am not getting 3.3V or atleat more than 3V? Do I need any external pullup or pulldown on that pin?

derFrickler, 2019-11-16:
Might be the optocouplers on the driver pulling to much current for the ESP32.

karoria, 2019-11-16:
Ok. Right now esp32 is using only computer power. Will it help if I add 5v external supply by smps?

karoria, 2019-11-16:
Ok. I have experimented by disconnecting all the drives. All drives disconnected gives 3.2V at pin. Then connecting each drive one after another drops 0.3V. So when I connect all the 3 drives, output becomes 2.3V. I have checked drive manual and it shows optocoupler at input, so @derFrickler your guess is right. Should I add a pull down resistor to maintain voltage level? If so, what can be possible effects in terms of delay? If at all it may result in missing pulses, then I will not use enable function.

derFrickler, 2019-11-16:
You will need to add a driver transistor to the ESP32 output as discussed here:
https://github.com/bdring/Grbl_Esp32/issues/217

karoria, 2019-11-16:
Ok. Will try that.

closed 2019-11-26 by karoria