bdring/Grbl_Esp32 #296: Feedhold and then resuming gives error 33

karoria, 2019-11-16:
Friends, sometimes when I use feedhold and then resume the program, it gives error. The screenshot is attached here. What can be the problem?
[image]

mac7988, 2019-11-17:
Hi Ravi,

This happens to me to man. I don't know what is going on. I just been ignoring it as it doesn't effect anything for me.

karoria, 2019-11-18:
Glad to know that somebody else is also in the same ride...;)

bdring, 2019-11-18:
Under what conditions does the error occur?

 - Where is the gcode coming from? (SD card, Gcode Sender, Manual entry, etc)
 - Where is the feedhold coming from? (WebUI, GCode Sender, Manual, etc)
 - Where is the resume coming from? (WebUI, GCode Sender, Manual, etc)

mac7988, 2019-11-18:
Hi Bart,

For menthe condition occurs when I run a gcode which contains a feed hold
command and I hit cycle start to resume I will get this error.

bdring, 2019-11-18:
@mac7988 

- Where is the gcode coming from? (SD card, Gcode Sender, Manual entry, etc)
- Where is the feedhold coming from? (WebUI, GCode Sender, Manual, etc)
 - Where is the resume coming from? (WebUI, GCode Sender, Manual, etc)

mac7988, 2019-11-18:
It doesn't matter manual a sender or SD card they both produce the same
error.

karoria, 2019-11-18:
Hi Barton,
In my case, I use cncjs for gcode sending, feedholding and resuming. I have also noticed the error occurs if i issue "!" and "~" from keyboard. My observation is, it generates error in the cases when i issued feedhold while program processed gcode lines with "i" and "j" values and resuming it.

bdring, 2019-11-18:
@karoria 

Can you attach a gcode file where you see the problem. 

Error 33 is associated with arcs G2 and G3

mac7988, 2019-11-19:
I was mistaken my error is see below
[image]

here is an example gcode I hand wrote.
G0 F1000
G90 G21 X0
G90 G21 X10
M8
G4 P1
M9
G90 G21 X0
M8
G4 P1
M9
G90 G21 X20
M8
G4 P1
M9
G90 G21 X40

G1 G90 G21 X0

M30

karoria, 2019-11-20:
Sorry for late response. My program is attached here.
[1036.txt](https://github.com/bdring/Grbl_Esp32/files/3868230/1036.txt)

bdring, 2019-11-23:
@karoria I ran your file from the SD card. I can hold and resume from both the WebUI and the Serial port. I did it about 20 times.

Is it possible some extra text is being sent while on hold? Are you doing anything while on hold?

karoria, 2019-11-24:
No extra text or no activity in hold mode. I don't use SD card. My USB cable is 1.5meter long (it is shielded and good quality). Can it be villain?