bdring/Grbl_Esp32 #297: Auto C-axis for plunge knife or drag knife CNCs

brakthehun, 2019-11-21:
**Please describe the feature you would like implemented**
When there is an XY movement,, I think it would be cool if there was an Auto C axis calculation based on the direction - atan2(#,#).
You would put in a $ setting for how many steps for a full revolution and GRBL would rotate the knife what ever direction the head is moving.
$ long int steps per revolution
$ boolean continuous rotation enable
$ boolean autocalculate C-Axis
$ long int plunge delay in milliseconds
$ int max angle before lift

Then it would need the $115,, $125 and ignore $105 and

**Why do you think this would improve Grbl_ESP32?**
I see this on textile machines and fancier CNC mills with oscillating knifes. I don't think the Arduino Uno has the processing power to fit this in.
**What do you need the feature for?**
Textile cutting,, laser diode output shape compensation,, Cricut Maker hacking,, 
**Will this feature appeal to a lot of users?**
Probably not,, but It's a cool feature and super useful.

bdring, 2019-11-21:
I was playing with this idea a few weeks ago. I had the A axis tracking the direction of travel. If the direction change was more than X degrees, it would left, turn, lower then move.

I was working, but the performance was poor. Circles, were very jerky. I think the planner was not happy.

I might work on it again in a few weeks.

brakthehun, 2019-11-28:
This is exactly what I'm hoping for. - Especially the lift cycle.