bdring/Grbl_Esp32 #300: External Stepper Pinout

Klinenator, 2019-12-02:
Is the pinout for external drivers the same as for the standard version of GRBL?

bdring, 2019-12-02:
What type of external drivers? If the drivers use opto isolators, it is unlikely the 3.3V ESP32 will light the LEDs. They are usually designed for 5V or higher.

I have successfully used those by wiring 5V to the plus side and putting a transistor/FET on the minus side.

bdring, 2019-12-02:
Here is a clip of that schematic.

[image]

karoria, 2019-12-02:
Sharing my experience here, I have successfully run DM542 and Easy Servo drives with 3.3V only (direct output of ESP32). So, I recommend to check with your driver. I think most of the industrial quality drives has threshold of approx 2.5V. So they run seamlessly.

Klinenator, 2019-12-02:
Im using the tb6600 steppers

cyberhasse, 2019-12-04:
Hi!
Im also using TB6600 and setting step pulse ($0=10) to 10 or 15 helped in my case..

Thank you for this great port of grbl :)
Regards.

orangezeroalpha, 2019-12-13:
I'd appreciate some comments along somewhat similar lines.  

I just purchased a few JMC IHSV57-30-18-36-01 which are integrated closed-loop DC servos but can use a pulse, direction, enable signal and can be used in the place of a stepper.  I was going to use linuxcnc, but I've also been wanting to use the esp32 for grbl, bcnc, etc.  

What I like:
-up to 48v DC connects directly to the motor itself (no need to have on the board)
-no need for stepper sockets on board
-no motor cable harness needed on board
-no external driver box for each motor, somewhat simpler cabling
 
I believe I'd need to use a logic level converter from 3.3v to 5v, but can I do any harm to try the 3.3v the esp32 typically sends?  I see the schematic on the motor has optoisolators.

Am I missing something?  Would five external drivers such as this work with grbl esp32?
Thank you.

bdring, 2019-12-13:
Optos have an LED in them that needs a current limiting resistor. The drive manufacturer does you a "favor" by including this resistor. They usually size it for 5v. 

Trying with 3.3v will not hurt anything, but if you are close to the threshold, you might miss a step without knowing it

closed 2020-01-10 by bdring