bdring/Grbl_Esp32 #302: Step invert -- step pulse bleed

jasonhillier, 2019-12-08:
Please answer the following questions.

What version of the firmware are you using?
```
commit b88d83f66e587ff65349630f6206f4d6e59b50b6
Merge: 11c8edc 137f9c2
Author: bdring <barton.dring@gmail.com>
Date:   Tue Nov 5 08:59:51 2019 -0600

    Merge pull request #283 from bdring/Devt
```

Is the problem repeatable?
yes

Under what conditions does the bug occur?
When step invert mask ($2=1) is set to true, on any axis

[image]

If I invert the step pin to idle at high in hardware, there is no bleed through ($2=0)

[image]

jasonhillier, 2019-12-08:
I've also written a simple sketch the pulses one pin or the other, and there is no bleed-through. So I am certain it is a software bug, not a hardware issue. Tested on multiple esp32-s to make sure.

terjeio, 2019-12-08:
It looks like a step inversion issue to me. A possible workaround is to enable USE_RMT_STEPS in config.h.

closed 2020-03-28 by bdring