bdring/Grbl_Esp32 #304: Physical buttons won't work

yasapl, 2019-12-10:
Hi, I'm using ver 1.1f on ESP32 30 pin Devkit board.

None of the physical buttons (feed hold, cycle start, reset and door) have any effect. All pins (GPIO34,35,36 and 39) with pull-up resistors 10k (to 3.3v)
Tested on two boards.
Limit switches / probe work fine, motors run without issues

Any idea? 

Thank you
Jacek

bdring, 2019-12-10:
Reply with the startup messages on your serial port.

Did you comment out `#define IGNORE_CONTROL_PINS`  in [config.h]?(https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/config.h#L284)

yasapl, 2019-12-10:
Of course I didn't. It is always something simple. 

Thank you very much, now buttons work as expected!

Startup message on serial port: 
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Using cpu_map...CPU_MAP_ESP32_DEVKITV1_30PIN]
Grbl 1.1f ['$' for help]

closed 2019-12-10 by bdring