bdring/Grbl_Esp32 #306: Feed hold does not stop feed when homing

owenduffy, 2019-12-16:
Please answer the following questions.

What version of the firmware are you using?

a7b9ffc

Is the problem repeatable?

Yes.

Under what conditions does the bug occur?

Homing.

**Important** If you paste firmware code, please use [Markdown Code and Syntax Highlighting](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) with language C++. Use the three back tick method.

```C++
  #define EASIER_TO_READ true
```
Feed hold is really useful to stop the machine if there is an obstruction in its path, but it does not work when homing. It would also be useful if for some reason the limit switch did not work properly.

mac7988, 2019-12-17:
> Please answer the following questions.
> 
> What version of the firmware are you using?
> 
> a7b9ffc
> 
> Is the problem repeatable?
> 
> Yes.
> 
> Under what conditions does the bug occur?
> 
> Homing.
> 
> Feed hold is really useful to stop the machine if there is an obstruction in its path, but it does not work when homing. It would also be useful if for some reason the limit switch did not work properly.

Hi Owen,
This is not a bug. It is working as intended. It is actually a request for enhancement.
Cheers!😃

bdring, 2019-12-17:
At this time there are no plans to change the homing sequence. I suggest using an E-Stop if there is an urgent need to stop the machine.

closed 2019-12-17 by bdring