bdring/Grbl_Esp32 #308: ESP module overheated. now in reboot cycle

kjjordans, 2020-01-01:
Please answer the following questions.

What version of the firmware are you using?
GRBL_VERSION "1.1f"
GRBL_VERSION_BUILD "20191208"
Is the problem repeatable?
don't know

Under what conditions does the bug occur?
always

I'm not sure it this is a bug or a problem with the dev board hardware. 
I have a ESP-WROOM-32 dev. wide dev board connected to the GRBL pen/laser V2 board

- initially had intermittent connection problems for ~10min
- then WIFI went down. 
- I tried to connect to the usb serial while connected to the machine without luck
- I removed the dev board from the carrier board, waited for it to cool down and connected to usb serial and got this repeating every half second or so. 
- the board was too hot to touch within about 1 min of connecting to usb 

_21:31:14.117 -> rst:0810 (RTCWDTORTCORESET),boot:0813 (SPIOFASTOFLASHOBOOT)
21:31:14.122 -> flash read err, 1000
21:31:14.122 -> etsOmain.c 371 
21:31:14.483 -> ets Jun  8 2016 00:22:57
21:31:14.483 -> 
21:31:14.483 -> rst:0x10 (RTCWDTORTCORESET),boot:0x13 (SPIOFASTOFLASHOBOOT)
21:31:14.483 -> flash read err, 1000
21:31:14.483 -> etsOmain.c 371 
21:31:14.853 -> ets Jun  8 2016 00:22:57
21:31:14.853 ->_ 

Any ideas of what might have gone wrong?

bdring, 2020-01-01:
It is most likely an ESP32 module problem.

Do you know where the heat  is coming from? Is it the ESP32 module or the linear regulator. Send a photo of where the hot spot is.

Try loading a simple example sketch.

ErikShorb, 2020-01-07:
I had an instance of this issue - unclear to me that this is necessarily a software bug.

Hardware:
 * New Grbl_ESP32_TMC2130_Plotter_Controller board.
 * NodeMCU 32s ESP32s board (original board provenance unremembered)
 * Bigtreetech TMC2130 v3 (configured for SPI from seller) : https://www.amazon.com/gp/product/B07RQ4819Z/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1
 * 12V 5A power supply

Scenario:
 * New install of Arduino IDE 1.8.10
 * Downloaded GRBL_ESP32
 * Compiled code, having changed nothing (thus configured in Demo mode)
 * Plugged in ESP32s board
 * Plugged in USB cable (connected to USB 3.0 hub)
 * Uploaded firmware 
 * Removed USB cable
 * installed 1 TMC2130 module in Y socket
 * installed 5v .8A stepper on Y
 * Plugged in 12v power supply (turned off)
 * Plugged in USB cable
 * TURNED ON 12v power

Board shortly began to reboot repeatedly - **touched D1 diode on ESP32s module and burning hot**.  
 * Quickly turned everything off.  
 * Waited till ESP32s module cooled 
 * removed TMC2130 module and stepper.  
 * Plugged in USB cable  and witnessed repeated reset (but no heating).
 * unable to load any other sketch (constant reset cycling)

Debugging (non-conclusive):
 * replaced ESP32s module with newly purchased one. (https://www.amazon.com/gp/product/B07QCP2451/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)
 * Plugged in USB cable
 * Uploaded firmware
 * Unplugged USB cable
 * Plugged in SAME TMC2130 module from initial trial.
 * Plugged in 12V power connector (power turned off)
 * Plugged in USB cable
 * Put my finger on the ESP32s module diode
 * turned on 12v power - Immediate heat on diode!  TURNED OFF power.
 * scratched head... read more info on getting started...

Made 2 changes at same time (so much for scientific method...)
1:
 * Configured firmware with CPU_MAP_TMC2130_PEN
 * loaded TMCstepper library
 * recompiled and uploaded new firmware

2: 
 * replaced TMC2130 module

I retried power and no thermal runaway.

**Replacing TMC2130 with original one and applying 12V caused immediate thermal runaway.**

Dilemma - unclear if original TMC2130 module was root cause, or collateral of initial configuration.

Some general thoughts (perhaps best added to GRBL_ESP32 board git readme's)
  * People should be reminded to have neither external power or stepsticks installed during initial power on and firmware upload (I had neither power on or stick installed) because one doesn't know what the manufacturer (or ones own previous use) left for firmware on the ESP32s module.
*  It's worth verifying that the Demo mode of the firmware doesn't leave any pins in a state that when connected to a stepstick it might cause a short to the ESP32.  
* My understanding is that for D1 on ESP32s module to be so hot, it would require some short from one or more ESP32s pin(s) to TMC2130 module.
* In the future, I'll always have my thumb on the ESP32s board when trying out new stepstick modules.  Removing power quickly prevented any obvious damage to the ESP32s.

Regrettably, having toasted 1 each ESP32s module and TMC2130 module, (and nearly toasting a second ESP32s module during debug) I'm reluctant to perform too many more experiments to isolate the issues. That said, magnifying glass examination of the (no non-functional) TMC2130 module shows no signs of solder bridge or other obvious damage.

Regards

misan, 2020-01-08:
Many H-bridges create a short when they release the magic smoke. I would
say that is the cause of your diode getting hot quickly.

Though H-bridges are protected to a short of the load they are delicate
creatures that die quickly if the coil wires are switched while it is on.

bdring, 2020-01-10:
I don't know what causes these problems. I have added the following note to the [compiling instructions on the wiki](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware).

**Note:** It is always best to program the ESP32 the first time unattached to a controller board. Any previous firmware could put pins in a state that damages the ESP32 when powered on. Program it unattached and verify the firmware via the USB first. Never plug in the ESP32 or any other items while powered on.

kjjordans, 2020-01-11:
Thanks everyone. It may have been a short, I have everything on a perf board at the moment. I'm working on a PCB which should make everything a bit more reliable.

closed 2020-01-11 by bdring