bdring/Grbl_Esp32 #311: esp32 aborts while running in lasermode

mvturnho, 2020-01-05:
Please answer the following questions.

What version of the firmware are you using?
latest master branch from github

Is the problem repeatable?
Yes when using gcode file from SD card or from laserGrbl telnet connection

Under what conditions does the bug occur?
While running the laser engraver from SD or telnet using lasergrbl

I got the code using:
`git clone https://github.com/bdring/Grbl_Esp32.git`

Then build it using platformIO in visual code.

did nothing to the cpu_map.h
and changed this in the config.h
```C++
#define CPU_MAP_ESP32

#define HOMING_CYCLE_0 (1<<X_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)

#define HOMING_FORCE_SET_ORIGIN
#define HOMING_FORCE_POSITIVE_SPACE

```

In the GRBL configuration using the webui I changed the following:

'''
Label | Value | Help
$11 | 2.010 | Junction   deviation, mm
$20 | 1 | Soft limits,   boolean
$21 | 1 | Hard limits,   boolean
$22 | 1 | Homing cycle,   boolean
$24 | 300 | Homing feed,   mm/min
$25 | 3000 | Homing seek,   mm/min
$32 | 1 | Laser mode,   boolean
$100 | 320 | X steps/mm
$101 | 160 | Y steps/mm
$110 | 8000 | X Max rate, mm/min
$111 | 8000 | Y Max rate, mm/min
$112 | 5000 | Z Max rate, mm/min
$120 | 200 | X Acceleration,   mm/sec^2
$121 | 200 | Y Acceleration,   mm/sec^2
$130 | 311   | X Max travel, mm
$131 | 340 | Y Max travel, mm
'''

After homing I start the laser and it rund for some lines, then after aprox 1 min it crashes.
Here the console log for two subsequent crashes
```
[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[MSG:HTTP Started]
[MSG:Using cpu_map...CPU_MAP_ESP32_V4]

Grbl 1.1f ['$' for help]
[MSG:Check Limits]
[MSG:'$H'|'$X' to unlock]
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1446 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x4008edbf on core 1

Backtrace: 0x400929a8:0x3ffbe880 0x40092bd9:0x3ffbe8a0 0x4008edbf:0x3ffbe8c0 0x400f9163:0x3ffbe900 0x400dfadd:0x3ffbe920 0x400dfc37:0x3ffbe940 0x400811f2:0x3ffbe960 0x40084dc9:0x3ffbe980 0x40082b11:0x3ffba170 0x4008761a:0x3ffba190 0x400831c3:0x3ffba1b0 0x4008f0f1:0x3ffba1d0

[... 141 lines cut ...]
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9

[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Using cpu_map...CPU_MAP_ESP32_V4]

Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found

```

bdring, 2020-01-05:
I am away from home for about a week. I cannot test this.  Can you try compiling with bluetooth and wifi disabled? Then run the file from the SD card via serial port commands.

That will help isolate the problem.

mvturnho, 2020-01-06:
No problem I will text this today and report back to you.

mvturnho, 2020-01-06:
Oké I tested this.
Without Bluetooth and wifi compiled the firmware and run the gcode from scratch.
That worked fine, it completed the whole file.
Then I run from lasergrbl over serial that also completed fine.

I recompiled the firmware with only wifi enabled and run the file from lasergrbl over serial and the esp32 stopped.

Hope this helps

JGJMatt, 2020-01-13:
I have the same problem.
Over serial it works 100% but with wifi it stops on both Lasergrbl and Laserweb.

mvturnho, 2020-02-01:
@bdring 
Can I help in any way solving this issue?

bdring, 2020-02-01:
When WiFi is on are you still sending via serial?

lhxhaxan, 2021-08-29:
@buildlog 
Hi. 
I have the same i ssuse by Laserweb via Telnet.
And I get this massage via serial:

```
--- Miniterm on COM2  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---    

[MSG:Grbl_ESP32 Ver 1.3a Date 20210816]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:PEN_LASER]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:Global stepper disable pin:GPIO(13)]
[MSG:X  Axis Standard Stepper Step:GPIO(12) Dir:GPIO(26) Disable:None Limits(-300.000,0.000)]
[MSG:Y  Axis Standard Stepper Step:GPIO(14) Dir:GPIO(25) Disable:None Limits(-300.000,0.000)]
[MSG:Laser spindle on Pin:GPIO(27), Enbl:None, Freq:5000Hz, Res:13bits Laser mode:On]

[MSG:Client Started]
[MSG:Connecting haxan_mi]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connected with 192.168.***.***]
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:X  Axis limit switch on pin GPIO(32)]
[MSG:Y  Axis limit switch on pin GPIO(4)]

Grbl 1.3a ['$' for help]
assertion "pbuf_free: p->ref > 0" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/pbuf.c", line 765, function: pbuf_free
abort() was called at PC 0x40159c7f on core 1

Backtrace: 0x40093b70:0x3ffd5260 0x40093da1:0x3ffd5280 0x40159c7f:0x3ffd52a0 0x4017c6df:0x3ffd52d0 0x40174c11:0x3ffd52f0 0x4017571d:0x3ffd5370 0x4017579a:0x3ffd53a0 0x400ec20e:0x3ffd53c0 0x4021fc2b:0x3ffd53f0 0x400e10f8:0x3ffd5410 0x400eb570:0x3ffd5840 0x400eb22c:0x3ffd5860 0x400da317:0x3ffd5880 0x400902b9:0x3ffd58a0
  #0  0x40093b70:0x3ffd5260 in reconfigureAllWdts at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x40093da1:0x3ffd5280 in commonErrorHandler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x40159c7f:0x3ffd52a0 in __kernel_rem_pio2 at /tmp/build/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libm/math/../../../.././newlib/libm/math/k_rem_pio2.c:274 (discriminator 3)
  #3  0x4017c6df:0x3ffd52d0 in netif_poll at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/netif.c:211
  #4  0x40174c11:0x3ffd52f0 in lwip_accept at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c:3532
  #5  0x4017571d:0x3ffd5370 in lwip_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c:3532
  #6  0x4017579a:0x3ffd53a0 in lwip_socket at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c:3532
  #7  0x400ec20e:0x3ffd53c0 in _GLOBAL__sub_I__ZN7SSwitchC2Etth at .pio\libdeps\release\TMCStepper\src\source/SERIAL_SWITCH.cpp:15
  #8  0x4021fc2b:0x3ffd53f0 in Spindles::YL620::set_speed_command(unsigned int, Spindles::VFD::ModbusCommand&) at Grbl_Esp32/src/Spindles/YL620Spindle.cpp:114
  #9  0x400e10f8:0x3ffd5410 in WebUI::NotificationsService::sendPushoverMSG(char const*, char const*) at Grbl_Esp32/src/WebUI/NotificationsService.cpp:157
  #10 0x400eb570:0x3ffd5840 in WebUI::WiFiConfig::ConnectSTA2AP() at Grbl_Esp32/src/WebUI/WifiConfig.cpp:179
  #11 0x400eb22c:0x3ffd5860 in String::operator+=(char const*) at Grbl_Esp32/src/WebUI/WifiConfig.cpp:179
      (inlined by) WebUI::WiFiConfig::info() at Grbl_Esp32/src/WebUI/WifiConfig.cpp:75
  #12 0x400da317:0x3ffd5880 in report_realtime_status(unsigned char) at Grbl_Esp32/src/Report.cpp:924
  #13 0x400902b9:0x3ffd58a0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57

```

and I didn`t  send anything when the gcode was running.
Is there any problem with telnet settings?

Thank you

bdring, 2021-08-29:
Post your machine definition

MitchBradley, 2021-08-29:
That is perhaps the strangest backtrace that I have ever seen.  Its elements of report_realtime_status(), WifiConfig::info(), WiFiConfig::ConnectSTA2AP(), NotificationsService::sendPushoverMSG(), YL620::set_speed_command(), TMCStepper SERIAL_SWITCH, and the lwip_socket chain are unrelated from the call chain standpoint.  I initially suspected a stack overflow, but the stack depth as indicated by the stack addresses in the backtrace seems to be about 0x640, which is not a lot.

lhxhaxan, 2021-08-30:
> Post your machine definition

I use Pen_Laser machine. Just change  x limit pin and SpindleType.

```C++
#pragma once
// clang-format off

/*
        pen_laser_custom.h
        Part of Grbl_ESP32

        Pen assignments for the Buildlog.net pen laser controller V1 & V2

        For pen mode be sure to uncomment #define USE_PEN_SERVO in config.h
        For solenoid  mode be sure to uncomment #define USE_PEN_SERVO in config.h
        For laser mode, you do not need to change anything
        Note: You can use all 3 modes at the same time if you want

        2019    - Bart Dring
        2020    - Mitch Bradley

        Grbl_ESP32 is free software: you can redistribute it and/or modify
        it under the terms of the GNU General Public License as published by
        the Free Software Foundation, either version 3 of the License, or
        (at your option) any later version.

        Grbl is distributed in the hope that it will be useful,
        but WITHOUT ANY WARRANTY; without even the implied warranty of
        MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
        GNU General Public License for more details.

        You should have received a copy of the GNU General Public License
        along with Grbl_ESP32.  If not, see <http://www.gnu.org/licenses/>.
*/

#define MACHINE_NAME "PEN_LASER"

// Pick a board version
//#define PEN_LASER_V1
#define PEN_LASER_CUSTOM

#define X_STEP_PIN              GPIO_NUM_12
#define X_DIRECTION_PIN         GPIO_NUM_26

#define Y_STEP_PIN              GPIO_NUM_14
#define Y_DIRECTION_PIN         GPIO_NUM_25

#define STEPPERS_DISABLE_PIN    GPIO_NUM_13

#ifdef PEN_LASER_V1
    #define X_LIMIT_PIN     GPIO_NUM_2
#endif
#ifdef PEN_LASER_V2
    #define X_LIMIT_PIN     GPIO_NUM_15
#endif
#ifdef PEN_LASER_CUSTOM
    #define X_LIMIT_PIN     GPIO_NUM_32
#endif
#define Y_LIMIT_PIN             GPIO_NUM_4

// #define USING_SERVO  // uncomment to use this feature

#ifdef USING_SERVO
    #define Z_SERVO_PIN             GPIO_NUM_27
#endif

#define SPINDLE_TYPE SpindleType::LASER
#define LASER_OUTPUT_PIN    GPIO_NUM_27

// defaults
#define DEFAULT_STEP_PULSE_MICROSECONDS 3
#define DEFAULT_STEPPER_IDLE_LOCK_TIME 250 // stay on

#define DEFAULT_STEPPING_INVERT_MASK 0 // uint8_t
#define DEFAULT_DIRECTION_INVERT_MASK 0 // uint8_t
#define DEFAULT_INVERT_ST_ENABLE 0 // boolean
#define DEFAULT_INVERT_LIMIT_PINS 1 // boolean
#define DEFAULT_INVERT_PROBE_PIN 0 // boolean

#define DEFAULT_STATUS_REPORT_MASK 1

#define DEFAULT_JUNCTION_DEVIATION 0.01 // mm
#define DEFAULT_ARC_TOLERANCE 0.002 // mm
#define DEFAULT_REPORT_INCHES 0 // false

#define DEFAULT_SOFT_LIMIT_ENABLE 0 // false
#define DEFAULT_HARD_LIMIT_ENABLE 0  // false

#define DEFAULT_HOMING_ENABLE 0
#define DEFAULT_HOMING_DIR_MASK 0 // move positive dir Z, negative X,Y
#define DEFAULT_HOMING_FEED_RATE 200.0 // mm/min
#define DEFAULT_HOMING_SEEK_RATE 1000.0 // mm/min
#define DEFAULT_HOMING_DEBOUNCE_DELAY 250 // msec (0-65k)
#define DEFAULT_HOMING_PULLOFF 3.0 // mm

#define DEFAULT_SPINDLE_RPM_MAX 1000.0 // rpm
#define DEFAULT_SPINDLE_RPM_MIN 0.0 // rpm

#define DEFAULT_LASER_MODE 0 // false

#define DEFAULT_X_STEPS_PER_MM 80.0
#define DEFAULT_Y_STEPS_PER_MM 80.0
#define DEFAULT_Z_STEPS_PER_MM 100.0 // This is percent in servo mode...used for calibration

#define DEFAULT_X_MAX_RATE 5000.0 // mm/min
#define DEFAULT_Y_MAX_RATE 5000.0 // mm/min
#define DEFAULT_Z_MAX_RATE 5000.0 // mm/min

#define DEFAULT_X_ACCELERATION 50.0 // mm/sec^2. 50 mm/sec^2 = 180000 mm/min^2
#define DEFAULT_Y_ACCELERATION 50.0 // mm/sec^2
#define DEFAULT_Z_ACCELERATION 50.0 // mm/sec^2

#define DEFAULT_X_MAX_TRAVEL 300.0 // mm NOTE: Must be a positive value.
#define DEFAULT_Y_MAX_TRAVEL 300.0 // mm NOTE: Must be a positive value.
#define DEFAULT_Z_MAX_TRAVEL 100.0 // This is percent in servo mode...used for calibration
```