bdring/Grbl_Esp32 #312: PCB to Test New Features

bdring, 2020-01-07:
I created a new test PCB to test some new features and hardware.

**External Stepper Drivers.**

There has been a lot of interest in powering external drivers. These are the types with opto isolators that run on 5V (~20ma). This PCB uses 74ACT245 chips to shift this voltage and can drive 50ma per channel. Most level shifters like TXS0101 cannot do this.

**New Spindle Types.**

This board will also test spindle control circuits to drive VFD type spindles. I have analog 0-10V and RS485/Modbus. 

It also has a ESC and 5V PWM via the 74ACT245. Some PWM speed controllers have a strong pull down resistor, which causes problems for some level shifters.

[image]

derFrickler, 2020-01-07:
Looks great, maybe you should consider some optos on the input pins for larger machines.

bdring, 2020-01-07:
That is probably a good idea. I used the EL817 before. I'll look at that one.

I have RC filtering on the inputs. That is probably not needed with them.

misan, 2020-01-07:
+1 @derFrickler suggestion.

Plus maybe, as there is room, an alternate power input with screw terminals ...

bdring, 2020-01-07:
@derFrickler The isolated power supply you need is always a pain.

@misan ... will add.

misan, 2020-01-08:
@bdring I understand 2.55mm DuPont terminals for I/O signals help keep the size (and cost) down. 

But the case for many users of external drivers and powered end-stops is they are going to have a bunch of wires to connect to the PCB. While it is possible to solder the proper connect for each set of signals, a screw terminal (or a press lever one, for example: https://www.aliexpress.com/item/10000004154911.html) would allow the easy connection of wires to the board without soldering. 

I have no trouble soldering but I know other users that avoid it at all costs and make a mess when they don't.

(My AI writing-style helper warns me my text reads like an angry post, this one is not :-)

This is just me providing feedback.

derFrickler, 2020-01-08:
You could use the power supply from the external stepper drivers for that. 
Even if you are using the same power supply used to power your board - no isolation - the optos will make the system more robust to external noise, as they require some current (20mA) and use an higher voltage. I used this board for exactly the same features on my arduino driven machine:
https://www.omc-stepperonline.com/de/5-axis-cnc-breakout-board-schnittstelle-mach3-cnc-router-kit.html

bdring, 2020-01-08:
@misan  I am not a fan of 2.54mm terminals. They are too small and I like to use ferrules.

@derFrickler If I don't use the isolated power supply I am sure I would get complaints from snarky people.

misan, 2020-01-09:
@bdring  +1 to use ferrules and larger screw terminals (not sure if that will set you off in terms of size/cost but definitely would be a welcome feature).

derFrickler, 2020-01-09:
@bdring  OK, or you could just use a separate input connector to power the optos, so its up to the user to connect whatever supply he wants.

bdring, 2020-01-09:
The isolated power supply is already on the board. It costs less than a dollar. 5V 200mA

[image]

misan, 2020-01-09:
@bdring Again, as there is room between the power input and the voltage regulator, a beefy electrolytic capacitor (that might be optional to add) can be welcome for cases where the power source is less than ideal.

bdring, 2020-01-10:
@misan Good idea, I already order 5 samples. If the board works out, I'll add it to the next revision. What value would you suggest?

misan, 2020-01-10:
@bdring Something that fits in the available space (not sure how much is there). This one (1000uF/35V) is 12.5 mm in diameter. https://www.amazon.com/Panasonic-FM-Aluminum-Electrolytic-Capacitors/dp/B076TZGGHS/

Or even better, (2200uF/35V) same diameter https://www.amazon.com/pcs-Panasonic-FR-Capacitors-2200uf/dp/B072SSJQSK

jschoch, 2020-01-12:
+1 for this specialized board and +1 terminal blocks over pin headers!

I would like to suggest that higher voltage range should be available on this board.  I think there are 2 camps in the US: The first will us a full VFD and 220.  The other camp will be using 110 and either DC brushed or hobby brushless grade spindles.  This is a good fit for pcb mills but 24v limits spindle speed quite a lot unless you have access to 220v.  If it is just a matter of the caps then maybe someone can easily make their own 48v capable board.

Wiring really is a nightmare on these small boards.  I milled a single sided board with 5.08 pitch terminal blocks and it is a very tight fit just breaking out the pins and there is no room for much else for a 70x100 board.   

[image]

The 2.54 spring terminals are a PITA when dense.  The format of the centroid acorn hardware seems to work pretty well and they have added nice touches like input LEDs.

I'd also like to mention that I am mid-build on my first JMC closed loop stepper project and those things put out a ton of noise.  Adding a VFD would make it worse.  I spent a few days trying different things and ran into issues with level shifting and GPIO12 voltage for reading boot from flash (guessing due to the current draw).  For the inputs I ended up adding RC filters and needing physical distance between the esp32 and the drivers.  The noise (1-2v swings between 2khz and 32khz) was there on the input pins without any wires connected and a few feet distance.  A good guide on dealing with noise will be helpful since there is so much less room to work when the input signals are 3.3v.  I wonder if 12 or 24v signals for the inputs which are then stepped down for the esp as close to the board reduces the chance for issues.   I also wonder about the pros/cons of RC vs LC vs opto when dealing with noise at 3.3v.  It appears that my JMC drivers did have opto isoltation so that may not have added anything of value.

Finally I wonder if rs232 control has any advantages for noise and the number of wires needed.  I can see advantages for VFD control for adding rs232.  my stepper drivers also support it but each implementation is likely different.

It would be great to have a larger thread to discuss reducing noise and tricks to optimize wiring.  Since I'm not sure where else to do it i'm hoping that it is ok to do here.

piotrpodlaski, 2020-01-23:
@bdring Firstly, great work in the project, I will be doing a custom ESP32 PCB to use it with your firmware for my CNC project.
Would you mind sharing the schematic of the board, I bought a cheap spindle from China with 0-10V interface that I would like to control the speed of. And another question, when do you plan to release 0-10V production in software? Is it there already on one of the development branches?

kripobs, 2020-02-10:
@bdring Hi there, any news on your test boards? :)

bdring, 2020-02-11:
@kripobs 

I have been so busy over the past few weeks. I did not get a chance to build it until yesterday. I found that I bought the wrong isolated power supply (lower center).

I can test all of the functions except for the optically isolated inputs. 

The (4) stepper motor output and 0-10V spindle control should be easy to test. The RS485 Modbus will be more work. I need to write the firmware and my VFD is 220V, which I don't have at my workstation.

[image]

kripobs, 2020-02-11:
Thanks for the update! I just finished building my CNC Router to a "beta-stage". I am using it for now with an uno and can not wait to replicate your board. Thank you for all your efforts!

yasapl, 2020-03-06:
Hi, is it only me or you did not include pins for hw buttons? Also you have double EN terminals and missed pulse ones (I guess this is only a print issue).

Any idea when can you make it available?

Thanks

bdring, 2020-03-06:
Yes, silkscreen error.

I need to finish testing the RS485 on the spindle. The 485 works,but need to test on the spindle. My spindle is 220V which I don't have in the shop. I borrowed a transformer, but have not wired it yet. 

On vacation for 2 weeks now. I am not sure when I can get some finished boards. The China supply chain is a mess.

yasapl, 2020-03-06:
thanks for quick reply. What about the hw buttons like cycle start / pause, stop, reset?

MitchBradley, 2020-03-06:
Looking at the cpu_map entry for that board, it appears that the pins that were previously used for such functions had to be repurposed as limits.  The A axis and modbus signals consumed several pins leaving none free for hw buttons.

You could reuse inputs like A limit or probe for hw buttons if you don't need them for their labeled purposes.

The Achille's heel of the ESP32 module is its shortage of pins; it doesn't have enough for everything you might want to do in a full-featured CNC system.  There are some proposals for using I/O expanders to fix that, but it will require some careful software work.

zrwd01, 2020-03-22:
waiting for new PCB layout

jschoch, 2020-03-27:
Looking forward to this board!

wondering about your thoughts on how to handle a fault signal from the driver.

bdring, 2020-03-27:
You would need to read the fault on an I/O pin.  You could do that with multiple I/O or use an OR gate.

Then you would need to decide what to do when a fault occurs. You could use existing hold and reset features.

Conversage, 2020-05-02:
this is exactly what i need! Cant wait for it to release.

vitalibr, 2020-05-29:
Amazing!

vitalibr, 2020-06-01:
@bdring I made a donation, it is little but that is what I can help now, my currency is 6 to 1 dollar. lol

This board is exactly what I need! Amazing idea and project! I was going to buy a mach3, but I prefer to use the grbl esp32 :)

Any predictions of her being released?

bdring, 2020-06-01:
Thank you for the donation. 

I don't know when I will release that board. It was primarily used to test some new firmware for spindles and the 74ACT245. We are spending all our time on adding additional I/O via I2S shift registers. [This board is being used for testing](https://github.com/bdring/6-Pack_CNC_Controller).

That can do everything the board mentioned in this thread and more.

bdring, 2020-06-01:
@vitalibr 

You can [join our Slack](https://join.slack.com/t/buildlog/shared_invite/zt-7twbithi-bRGScOaCgJoBpTbkykfxoA) if you want to help develop Grbl_ESP32.

wayneosdias, 2020-06-03:
Hey Bart,
I donated and bought your Tindie TMC board to help support the cause, Thanks. I wish this breakout board was available vs the driver board as I use powered servos. Im glad youve moved to screw terminals vs frictionlock/header types for robust connections. Anyways great project and support, get that board done and let us know when its available :)

atlaste, 2020-07-12:
@bdring Looks like you crammed all the stuff I designed myself on a nicely looking professional board. I would definitely buy this, as I'm struggling hard with my own messy solution. Just let me know when...

bdring, 2020-07-12:
This is the new hotness.

All the new firmware and hardware features perfected over the last few month are here.

[Github Repo](https://github.com/bdring/6-Pack_CNC_Controller)

[Here is a draft of a blog/video post describing it.](https://docs.google.com/document/d/1a__PgSNCgN9fd9TaXW7yoGHKFkZJp84snypWQ6qfh4g/edit?usp=sharing) I just need to shoot some video next week. 

Production hardware is ordered

[image]