bdring/Grbl_Esp32 #314: ESPduino-32 with GRBL shield reboots over and over

ipsod, 2020-01-08:
Hi.  This is a little off topic, as I'm trying to control steppers for a not-grbl-related project, but this project seemed like the place to find people who would have an answer.

I'm trying to use an ESPduino-32 with a grbl shield.  I'm compiling and uploading code using arduino-ide.  The ESP32 works as expected until I plug the shield in (without any stepper drivers, power, or anything else plugged in to the shield), and it starts rebooting over and over.

I think I saw that some people here are using these boards and these shields.  What am I doing wrong?

derFrickler, 2020-01-08:
If i remember right the problem is the stepper enable pullup resistor on the shield. 
This conflicts with one of the ESP32 boot pins. 
Try unsoldering the resistor on the stepper shield.

ipsod, 2020-01-08:
That seems to have done the trick.  Thank you!

closed 2020-01-08 by ipsod