bdring/Grbl_Esp32 #316: Wiki documentation suggestions

mfarabee, 2020-01-10:
I hope this is the right way to suggest Wiki (documentation) updates.

Here are few suggestions based on my working with the software:

1) I did not know understand about the Flash settings. I did my first compile and upload and it worked great. I then tried to recompile and reset some of the defaults. I did not realize that many of the settings were stored in flash (non volatile) and would not be reset unless the SETTING_VERSION changes. I think an explanation of this is useful. Also I initially struggled getting connected via the LAN. Since I was connected to my computer after downloading the code, I was expecting to be able to upload the web server from there.   It almost feels like the default RADIO_MODE should be STA and I can setup WiFi or Bluetooth once I have everything running and out of test mode.

2) I ran into a problem on how to stop Bluetooth and go back to AP mode.  I would recommend putting a note in the documentation stating that you can execute [ESP110]AP in the command window. Another great option, if you are using "Android Controller" app, I recommend creating a custom button to allow quick switching back to AP mode.  Settings->Jogging preferences->Custom buttons -> create AP = {ESP110]AP

3) I was a little confuse about what stepper defaults to set. I saw in the defaults.h that there were settings for microsteps. I was confused how to set this up since I was using DRV8825 drivers. It finally hit me that this is only for "SPI programable Drivers". It might be good to have a section in the Wiki that explains what is needed to setup for Trimanic, Stepstick, solenoid, etc. 

4) I would recommend a Wiki page something like "Getting Started". From here I would give a brief overview of where to look for things and what you need to edit to get started.

-  I would point people to the GRBL docs for additional information as much of it is still valid..

- There are two types of setup data ESP32 configuration: wifi/Bluetooth,etc… and hardware configuration: stepper , gcode, etc. The ESP32 configuration data is stored as persistent data and can be set by issuing [ESPXXX] commands after compilation. The initial values for these commands can be found .at ….
A list of the [ESPXXX] command can be found at doc/Command.txt. Current settings can be found by issuing [ESP400] from the command line..

- The Hardware configuration command are also persistent and can be set by issuing $xxx commands after compilation. The initial default values are stored "defaults.h" file. A list of "$" command can be found by issuing "$$" from command line.

Files to edit:
config.h - defines which pin configuration to used (see cpu_map.h). Change from CPU_TEST_DRIVE to CPU_MAP_ESP32 (or some other dedicated CPU map) to activate steppers.
cpu_map.h - defines pin configurations for development boards. This is where to find existing pin configurations or create/modify pin configurations.
defaults.h - Initial default values for hardware configuration.
settings.h - Change SETTINGS_VERSION to force reset of flash.

Great job and thank you!

bdring, 2020-01-11:
@mfarabee 
**Thanks for the input.** I will add your suggestions soon. I know the wiki is lacking in many areas and needs to be re-organized. 

I have been working on a comprehensive "CPU maps for dummies". While writing, I found that a few tweaks to the firmware could simplify things. I am implementing those soon.

After that, I will publish the page and work on some of your suggestions. The organization is as important as the content, so suggestions on where to put your content is helpful as well.

bdring, 2020-01-11:
I updated the [Bluetooth page](https://github.com/bdring/Grbl_Esp32/wiki/Using-Bluetooth) with some of your suggestions.

The custom button idea is great, but I don't think that should be part of the doc.

If you have suggestions, to that page, please let me know.

mfarabee, 2020-01-11:
I think the Bluetooth addition is great. I can understand not wanting to put app setup instructions in your docs.

By the way I was very impressed with the amount of commenting and documentation within the code. I was able to dig into the code and figure many things out. The code comments were instrumental in being able to do this. In fact, the existing Wiki is awesome as well. 

My suggestions are more for tying it all together, especially for non programmers that may have a hard time traversing the code.  Some of it can be confusing like steppers or endstops on what and where to make the setup/edits. Some of it may also be having the right mind set. Is it better to compile with existing defaults and the change them in the GUI or is it better to code the defaults prior to upload? My initial thought (being a programmer) was to have all the defaults defined in the code, but now I realize that setting them later is perfectly fine. Having the Wiki describe the installation and setup process (what to do in code verses in GUI) would have been helpful.

Great work and I am look forward to ordering a board when they are available!