bdring/Grbl_Esp32 #319: Using a single limit switch pin for all axis

derFrickler, 2020-01-18:
**Please describe the feature you would like implemented**
As Pins are quite limited on the ESP32 it could make sense to only use one pin as limit switch input for all axis. this would free up some pins for other things, especially on more axis machines.
Used as limit switch, there should be no difference, the machine will halt if it is fired. 
Used as homing switch, the homing procedure has to home one axis by another to know which axis hit the limit.

**Why do you think this would improve Grbl_ESP32?**
It would free up pins for buttons, outputs etc..especially if all 6 axis are used.

**What do you need the feature for?**
want to play around with a 6 axis board and this is where the pins get problematic.

**Will this feature appear to a lot of users?**
Might be, if they need more free pins for other stuff - maybe it could free up pins for a small display.

bdring, 2020-01-20:
The main issue to deal with is what to do if a limit is triggered before homing. Typically it would attempt to back off the axis that it is trying to home. That might not be the axis that that is triggering. You could back every axis off, but that seems a little sloppy. The other option is to fail to home and force the user to fix the problem.

derFrickler, 2020-01-20:
I don't really understand. 
If you are homing one axis after another, only one axis is moving at a time, so only one can hit the switch. That that one can then be backed off.

bdring, 2020-01-20:
If a limit switch is already triggered before you home. Which one is it?

bdring, 2020-01-20:
Just for fun, I tried a cpu map with....

#define X_LIMIT_PIN     	GPIO_NUM_17
#define Y_LIMIT_PIN     	GPIO_NUM_17
#define Z_LIMIT_PIN     	GPIO_NUM_17

It appears to work fine. If you send ? it will report all axes triggered from one switch.

If a switch is being pushed by an axis during homing, which is not the homing axis, it will back off the homing axis, which will not clear the switch and it will error. You can then manually investigate the problem and fix it.

bdring, 2020-01-20:
Note: you should probably should make sure only one axis homes at a time.

derFrickler, 2020-01-21:
Great!  Thanks for the efforts of testing it. Had not thought it was that easy.
That is exactly the way it works on my other machine with ESTLCam.

P.S. The problem of which switch triggered exists even with separate switches per axis if you have a switch on each side. the machine will not know which limit of the axis has been reached ;_)

closed 2020-01-21 by derFrickler

bdring, 2020-01-21:
There is a setting for that. You might experiment with it. I just want to remind you of the problem of multiple switches.

https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/config.h#L221