bdring/Grbl_Esp32 #322: What is the maximum file size supported in SD card mode

karoria, 2020-01-22:
Hi @bdring 
Thanks for this wonderful port of grbl! I am curious about how big the file it can handle via SD card mode (and also the largest capacity of SD card supported) if I use web interface. Are there any known bugs? 
I am asking all this because I already use grbl_esp32 with USB serial communicated with RasPi4 with cncjs software and I am facing issues regarding update of coordinates for larger files say more than 4MB. I am thinking to use SD and web interface now. My machine is CNC milling. Can I use it with my machine? Some of my programs are as large as 40MB.
Can I use pen drive instead of SD card by changing pcb footprint to usb pen? or it is not that easy?

bdring, 2020-01-22:
40MB is no problem, What is a pen drive? Do you mean USB flash drive? No.

closed 2020-01-22 by bdring

karoria, 2020-01-22:
Yes, I meant USB flash drive. So it is not possible. Can I use 64GB class 10 micro SD ? What is the max limit of SD that can be used?

bdring, 2020-01-22:
It uses the Arduino SD library. The max size is 32GB.  It supports SD and SDHC type cards. They should be formatted in FAT16 or FAT32.

karoria, 2020-01-22:
Ok. Got it. I will give it a try.