bdring/Grbl_Esp32 #324: WDT forces reset while running

maudette01432, 2020-01-30:
During some cuts while connected over serial the watchdog timer resets the device:

>>> X460.469Y17.461
ok
>>> X461.681Y24.710
ok
>>> X461.699Y24.902
E (1481834) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (1481834) task_wdt:  - IDLE0 (CPU 0)
E (1481834) task_wdt: Tasks currently running:
E (1481834) task_wdt: CPU 0: serialCheckTask
E (1481834) task_wdt: CPU 1: loopTask
E (1481834) task_wdt: Aborting.
abort() was called at PC 0x400fbe5c on core 0
Backtrace: 0x400929a8:0x3ffbe320 0x40092bd9:0x3ffbe340 0x400fbe5c:0x3ffbe360 0x40084dc9:0x3ffbe380 0x4000bfed:0x3ffd2940 0x4009025d:0x3ffd2950 0x4008ee4b:0x3ffd2970 0x40186efa:0x3ffd29b0 0x40175288:0x3ffd29d0 0x400eaae5:0x3ffd29f0 0x400eab61:0x3ffd2a20 0x400eac05:0x3ffd2a40 0x400eacc9:0x3ffd2a60 0x4021b48a:0x3ffd2a80 0x400f8526:0x3ffd2ab0 0x400f85c1:0x3ffd2ad0 0x400efedf:0x3ffd2af0 0x400f1c5d:0x3ffd2d10 0x400e262d:0x3ffd2d60 0x400e9d2a:0x3ffd2da0 0x400e995c:0x3ffd2dc0 0x400df065:0x3ffd2de0 0x4008f0f1:0x3ffd2e00
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Axis count 3]
[MSG:RMT Steps]
[Error] An error was detected while sending 'X456.727Y9.900': Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled. Streaming has been paused.
>>> X461.744Y77.488
Core 1 register dump:
PC      : 0x4008120a  PS      : 0x00060031  A0      : 0x40084dcc  A1      : 0x3ffbe960  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc69d8  A5      : 0x3ffc5a0c  
A6      : 0x00000000  A7      : 0x3ffbaaf4  A8      : 0x80081202  A9      : 0x3ffbe940  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x800925ef  A13     : 0x3ffd0c80  
A14     : 0x3ffb9a20  A15     : 0xbaad5678  SAR     : 0x00000005  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000011  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  
Core 1 was running in ISR context:
EPC1    : 0x4008120a  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40087601
Backtrace: 0x4008120a:0x3ffbe960 0x40084dc9:0x3ffbe980 0x4000bfed:0x3ffd0cd0 0x4009025d:0x3ffd0ce0 0x400fdcc2:0x3ffd0d00 0x40189f62:0x3ffd0d40 0x400e079d:0x3ffd0d70 0x400d2979:0x3ffd0db0 0x400f99b7:0x3ffd0dd0 0x4008f0f1:0x3ffd0df0
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Client Started]
[MSG:Connecting Restrictedayerspace Brewing]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connected with 192.168.1.61]
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Spindle init on pin 2]
[MSG:Using cpu_map...CPU_MAP_ESP32_V4]
Grbl 1.1f ['$' for help]

GCODE: "Stage2BottomPlate.nc.txt"

[Stage2BottomPlate.nc.txt](https://github.com/bdring/Grbl_Esp32/files/4131744/Stage2BottomPlate.nc.txt)

**Machine Configuration ($$):**
>>> $$
[MSG:Extended 0]
$0 = 3    (Step pulse time, microseconds)
$1 = 250    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 200.000    (Homing locate feed rate, mm/min)
$25 = 2000.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000.000    (Maximum spindle speed, RPM)
$31 = 0.000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 250.000    (X-axis travel resolution, step/mm)
$101 = 250.000    (Y-axis travel resolution, step/mm)
$102 = 800.000    (Z-axis travel resolution, step/mm)
$110 = 5000.000    (X-axis maximum rate, mm/min)
$111 = 5000.000    (Y-axis maximum rate, mm/min)
$112 = 5000.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 200.000    (Z-axis acceleration, mm/sec^2)
$130 = 300.000    (X-axis maximum travel, millimeters)
$131 = 300.000    (Y-axis maximum travel, millimeters)
$132 = 300.000    (Z-axis maximum travel, millimeters)

bdring, 2020-01-30:
How are you sending the gcode?

@luc-github Do you see anything with the netwrok settings?

luc-github, 2020-01-30:
without decoded stack - no idea sorry

bdring, 2020-01-30:
@maudette01432 Try pasting that backtrace into the exception decoder.

https://github.com/me-no-dev/EspExceptionDecoder

It will give us the exact location the error occured and all the calling functions.

maudette01432, 2020-01-31:
Interestingly..it seems to be time related.

I am sending with UGS (Platform) Nightly Build on OS X via serial @ 115200.

I ran a couple other tests today... After each reset there is an entry that looks like time - and it was the same both times (i.e. after x minutes reset):

**** Finished sending file in 00:15:33 ****

E (1499659) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (1499659) task_wdt:  - IDLE0 (CPU 0)
E (1499659) task_wdt: Tasks currently running:
E (1499659) task_wdt: CPU 0: serialCheckTask
E (1499659) task_wdt: CPU 1: loopTask
E (1499659) task_wdt: Aborting.
abort() was called at PC 0x400fbe5c on core 0
Backtrace: 0x400929a8:0x3ffbe320 0x40092bd9:0x3ffbe340 0x400fbe5c:0x3ffbe360 0x40084dc9:0x3ffbe380 0x4000bfed:0x3ffd2a80 0x4009025d:0x3ffd2a90 0x4008ee4b:0x3ffd2ab0 0x40186efa:0x3ffd2af0 0x401752c5:0x3ffd2b10 0x400eaae5:0x3ffd2b30 0x400eab61:0x3ffd2b60 0x400eac05:0x3ffd2b80 0x400eacc9:0x3ffd2ba0 0x4021b48a:0x3ffd2bc0 0x400f8526:0x3ffd2bf0 0x400f85c1:0x3ffd2c10 0x400efedf:0x3ffd2c30 0x400f1c5d:0x3ffd2e50 0x400e262d:0x3ffd2ea0 0x400e9d2a:0x3ffd2ee0 0x400e995c:0x3ffd2f00 0x400df065:0x3ffd2f20 0x4008f0f1:0x3ffd2f40
Rebooting...
ets Jun  8 2016 00:22:57

Send again...wait jog around a bit

>>> $J=G20G91X0.007F4
ok
>>> $J=G20G91X0.007F4
ok

**** Canceling file transfer. ****

E (1499581) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (1499581) task_wdt:  - IDLE0 (CPU 0)
E (1499581) task_wdt: Tasks currently running:
E (1499581) task_wdt: CPU 0: wifi
E (1499581) task_wdt: CPU 1: loopTask
E (1499581) task_wdt: Aborting.
abort() was called at PC 0x400fbe5c on core 0
Backtrace: 0x400929a8:0x3ffbe320 0x40092bd9:0x3ffbe340 0x400fbe5c:0x3ffbe360 0x40084dc9:0x3ffbe380 0x40148f51:0x3ffd6dd0 0x40148fc8:0x3ffd6e20 0x40213571:0x3ffd6e40 0x402136b0:0x3ffd6e60 0x40216771:0x3ffd6e80 0x4021680d:0x3ffd6ea0 0x40095add:0x3ffd6ec0 0x4008f0f1:0x3ffd6f00
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

maudette01432, 2020-01-31:
I left it sitting there...and sure enough after some amount of time it looks like it reboots - Stack Trace...

0x400eaae5: WiFiClientRxBuffer::r_available() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 49
0x400eab61: WiFiClientRxBuffer::fillBuffer() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 71
0x400eac05: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 107
0x400eacc9: WiFiClient::read(unsigned char*, unsigned int) at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 434
0x4021b48a: WiFiClient::read() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 345
0x400f8526: Stream::timedRead() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Stream.cpp line 36
0x400f85c1: Stream::readStringUntil(char) at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Stream.cpp line 287
0x400efedf: WebServer::_parseRequest(WiFiClient&) at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WebServer/src/Parsing.cpp line 70
0x400f1c5d: WebServer::handleClient() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WebServer/src/WebServer.cpp line 304
0x400e262d: Web_Server::handle() at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_153144/sketch/web_server.cpp line 1637
0x400e9d2a: WiFiServices::handle() at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_153144/sketch/wifiservices.cpp line 164
0x400e995c: WiFiConfig::handle() at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_153144/sketch/wificonfig.cpp line 551
0x400df065: serialCheckTask(void*) at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_153144/sketch/serial.cpp line 191
0x4008f0f1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
0x400929a8: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x40092bd9: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400fbe5c: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 174
0x40090026: xTaskGetSchedulerState at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4013
0x4008ea44: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 724
0x40186f12: sys_mutex_unlock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 93
0x4017531c: lwip_ioctl_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c line 3491
0x400eaae5: WiFiClientRxBuffer::r_available() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 49
0x400eab61: WiFiClientRxBuffer::fillBuffer() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 71
0x400eac05: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 107
0x400eacc9: WiFiClient::read(unsigned char*, unsigned int) at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 434
0x4021b48a: WiFiClient::read() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 345
0x400f8526: Stream::timedRead() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Stream.cpp line 36
0x400f85c1: Stream::readStringUntil(char) at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Stream.cpp line 287
0x400efedf: WebServer::_parseRequest(WiFiClient&) at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WebServer/src/Parsing.cpp line 70
0x400f1c5d: WebServer::handleClient() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WebServer/src/WebServer.cpp line 304
0x400e262d: Web_Server::handle() at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_153144/sketch/web_server.cpp line 1637
0x400e9d2a: WiFiServices::handle() at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_153144/sketch/wifiservices.cpp line 164
0x400e995c: WiFiConfig::handle() at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_153144/sketch/wificonfig.cpp line 551
0x400df065: serialCheckTask(void*) at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_153144/sketch/serial.cpp line 191
0x4008f0f1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

** THIS IS FOR THE ONE THAT WAS JUST SITTING THERE IDLE

maudette01432, 2020-01-31:
For the really short one:

Decoding stack results
0x400929a8: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x40092bd9: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400fbe5c: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 174
0x4008f0f1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

luc-github, 2020-01-31:
sorry, can you describe what actions you are doing when it crash?
you wrote it crash when doing serial but there no decoded stack of this crash.
then you wrote it crashed when idle but stack show there is wifi streaming when it  crash which should not happen if idle

maudette01432, 2020-02-10:
luc-github - I added a couple of the stack traces above.  The scenario where it rebooted was it was plugged into my iMac and connected to UGS (platform version nightly build).  It was simply sitting there for several minutes after jogging it around a bit.  In another situation (also connected via serial) while sending Gcode to make a cut it crashed and rebooted.  I will try the latest code (updated 4 days ago) to see if it helps anything as it looks like there was some major re-write of messaging code.

maudette01432, 2020-02-11:
I reflashed with the latest code and am connected only via WiFi with a 19.5V 4.5A power supply.  I have this GCODE which I am trying to run with CPU_MAP_ESP32 (no steppers installed in sockets).  Rest of the config as cloned.
[Thistle_in_dish_3.5x3.5.gcode.zip](https://github.com/bdring/Grbl_Esp32/files/4188507/Thistle_in_dish_3.5x3.5.gcode.zip)

<Run|MPos:80.636,39.044,-2.795|FS:595,1000|WCO:50.000,50.000,0.000|SD:35.01,/ASPIRE/Thistle_in_dish_3.5x3.5.gcode>
<Run|MPos:92.444,39.364,-2.207|FS:169,1000|SD:35.08,/ASPIRE/Thistle_in_dish_3.5x3.5.gcode>
<Run|MPos:66.392,39.364,-3.800|FS:458,1000|SD:35.20,/ASPIRE/Thistle_in_dish_3.5x3.5.gcode>
<Run|MPos:48.428,39.364,-4.316|FS:402,1000|SD:35.30,/ASPIRE/Thistle_in_dish_3.5x3.5.gcode>
<Run|MPos:28.216,39.364,-3.540|FS:929,1000|SD:35.44,/ASPIRE/Thistle_in_dish_3.5x3.5.gcode>
<Run|MPos:7.392,39.364,-2.193|FS:115,1000|SD:35.53,/ASPIRE/Thistle_in_dish_3.5x3.5.gcode>
<Run|MPos:24.904,39.680,-3.215|FS:786,1000|SD:35.62,/ASPIRE/Thistle_in_dish_3.5x3.5.gcode>
**Connection error**
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:50.000,50.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>

I can see of no way to get a stack trace when connected only via the web page and the "**Conection error**" message doesn't tell me anything - other than at ~35.62% this crashed.  Does ESP32 GRBL require a constant web connection to run through a GCODE file to completion?  Also I ran the file a second time and it progressed past the point where I got the message.  I know this Gcode is good because I have run it on an Arduino GRBL 1.1 based board and cut it on a machine.  It appears to be a random reset/connection issue where it just reboots.  I've tried a couple different ESP-32's I have from different makers and they all seem to randomly give connection errors.  Nothing appears to reproduce the problem in a consistent manner.

A subsquent run of the same code crashed elsewhere:

<Run|MPos:55.096,72.384,-3.023|FS:276,1000|SD:85.06,/ASPIRE/Thistle_in_dish_3.5x3.5.gcode>
<Run|MPos:43.596,72.384,-4.566|FS:418,1000|SD:85.26,/ASPIRE/Thistle_in_dish_3.5x3.5.gcode>
Connection error
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:50.000,50.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>

Because I can't get through a single code without some random crash.  How do you propose I capture the random connection errors?

bdring, 2020-02-12:
You must be connected to via USB/Serial port to get additional information during bootup and crashes. You can be also connected via Bluetooth or WiFi, but much of the data will only be available on the USB/Serial port.

[I have written an explanation on the wiki](https://github.com/bdring/Grbl_Esp32/wiki/FAQ#submitting-bug-information).

That information is very helpful for us if we cannot repeat the error.

maudette01432, 2020-02-12:
I got another crash with the same GCODE in USGS via serial...again at a different spot.  

>>> G1X-1.7338Y-0.0313Z-0.0855
ok
>>> G1X-1.7045Z-0.0900
E (2404887) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:

Decoding stack results
0x400929a8: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x40092bd9: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400fc058: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 174
0x4009025d: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4274
0x400929a8: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008ead5: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 828
0x40092bd9: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400fc058: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 174
0x4018710e: sys_mutex_unlock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 93
0x401873e6: sys_arch_unprotect at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 483
0x40174de5: lwip_ioctl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c line 3133
0x4009025d: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4274
0x4008ead5: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 828
0x4018710e: sys_mutex_unlock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 93
0x401873e6: sys_arch_unprotect at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 483
0x40174de5: lwip_ioctl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c line 3133
0x401754b9: lwip_ioctl_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c line 3490
0x400eacd1: WiFiClientRxBuffer::r_available() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 49
0x400ead4d: WiFiClientRxBuffer::fillBuffer() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 71
0x400eadf1: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 107
0x400eaeb5: WiFiClient::read(unsigned char*, unsigned int) at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 434
0x4021b6fe: WiFiClient::read() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 345
0x400f8722: Stream::timedRead() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Stream.cpp line 36
0x400f87bd: Stream::readStringUntil(char) at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Stream.cpp line 287
0x400f00d7: WebServer::_parseRequest(WiFiClient&) at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WebServer/src/Parsing.cpp line 70
0x400f1e55: WebServer::handleClient() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WebServer/src/WebServer.cpp line 304
0x400e2819: Web_Server::handle() at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_312146/sketch/web_server.cpp line 1637
0x400e9f16: WiFiServices::handle() at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_312146/sketch/wifiservices.cpp line 164
0x400e9b48: WiFiConfig::handle() at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_312146/sketch/wificonfig.cpp line 551
0x400df25c: serialCheckTask(void*) at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_312146/sketch/serial.cpp line 163
0x4008f0f1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
0x401754b9: lwip_ioctl_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c line 3490
0x400eacd1: WiFiClientRxBuffer::r_available() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 49
0x400ead4d: WiFiClientRxBuffer::fillBuffer() at /Users/maudette01432/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiClient.cpp line 71
[... cut ...]
0x400e9f16: WiFiServices::handle() at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_312146/sketch/wifiservices.cpp line 164
0x400e9b48: WiFiConfig::handle() at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_312146/sketch/wificonfig.cpp line 551
0x400df25c: serialCheckTask(void*) at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_312146/sketch/serial.cpp line 163
0x4008f0f1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

maudette01432, 2020-02-12:
Another run this morning (same gcode) different crash.  This time I compiled without BT and turned error messages on:

>>> G1X-0.7849Z-0.1553
E (1392589) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (1392589) task_wdt:  - IDLE0 (CPU 0)
E (1392589) task_wdt: Tasks currently running:
E (1392589) task_wdt: CPU 0: serialCheckTask
E (1392589) task_wdt: CPU 1: loopTask
E (1392589) task_wdt: Aborting.
abort() was called at PC 0x400fcdac on core 0
Backtrace: 0x4008ce9c:0x3ffbe170 0x4008d0cd:0x3ffbe190 0x400fcdac:0x3ffbe1b0 0x40084d55:0x3ffbe1d0 0x4000bfed:0x3ffb3ad0 0x4008a751:0x3ffb3ae0 0x40089021:0x3ffb3b00 0x40182e56:0x3ffb3b40 0x4018312a:0x3ffb3b60 0x40170ca5:0x3ffb3b80 0x40171379:0x3ffb3ba0 0x400eb7f9:0x3ffb3bc0 0x400eb861:0x3ffb3bf0 0x400eb8cd:0x3ffb3c10 0x400ebd21:0x3ffb3c30 0x4019aa0a:0x3ffb3c60 0x400f9e06:0x3ffb3c90 0x400f9ea1:0x3ffb3cb0 0x400f16df:0x3ffb3cd0 0x400f3541:0x3ffb3ef0 0x400e1f39:0x3ffb3f40 0x400eb168:0x3ffb3f80 0x400eaf64:0x3ffb3fa0 0x400df287:0x3ffb3fc0 0x400895e5:0x3ffb3fe0
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using cpu_map:CPU_MAP_ESP32_V4]
[MSG:Axis count 3]
[MSG:RMT Steps]
[Error] An error was detected while sending 'G1X-0.9397Z-0.1465': Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled. Streaming has been paused.
>>> G1X-0.6928Z-0.1560
Core 1 register dump:
PC      : 0x40081196  PS      : 0x00060031  A0      : 0x40084d58  A1      : 0x3ffbe7b0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc2120  A5      : 0x3ffc1130  
A6      : 0x00000000  A7      : 0x3ffb83a4  A8      : 0x8008118e  A9      : 0x3ffbe790  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x8008cae3  A13     : 0x3ffb1e60  
A14     : 0x3ffb8000  A15     : 0xbaad5678  SAR     : 0x00000005  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000011  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  
Core 1 was running in ISR context:
EPC1    : 0x40081196  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400873d9
Backtrace: 0x40081196:0x3ffbe7b0 0x40084d55:0x3ffbe7d0 0x4000bfed:0x3ffb1eb0 0x4008a751:0x3ffb1ec0 0x400febf2:0x3ffb1ee0 0x40185e62:0x3ffb1f20 0x400e0995:0x3ffb1f50 0x400d25b6:0x3ffb1f90 0x400fb25b:0x3ffb1fb0 0x400895e5:0x3ffb1fd0
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using cpu_map:CPU_MAP_ESP32_V4]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Client Started]
[MSG:Connecting Restrictedayerspace Brewing]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connected with 192.168.1.61]
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
Grbl 1.1f ['$' for help]
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: ADMIN_PWD NOT_FOUND
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: USER_PWD NOT_FOUND

Stack Trace output:

Decoding stack results
0x40081196: onStepperDriverTimer(void*) at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_312146/sketch/stepper.cpp line 296
0x40081196: onStepperDriverTimer(void*) at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_312146/sketch/stepper.cpp line 296
0x4008a751: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4274
0x4008a751: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4274
0x400febf2: esp_intr_alloc_intrstatus at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/intr_alloc.c line 666
0x400febf2: esp_intr_alloc_intrstatus at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/intr_alloc.c line 666
0x40185e62: timer_isr_register at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/timer.c line 206
0x40185e62: timer_isr_register at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/timer.c line 206
0x400e0995: stepper_init() at /var/folders/cl/fdfr2ycs6596pr7y3f2wrws40000gn/T/arduino_build_312146/sketch/stepper.cpp line 538
0x400d25b6: setup() at /Users/maudette01432/sandbox/Grbl_Esp32/Grbl_Esp32/Grbl_Esp32.ino

bdring, 2020-02-12:
Thanks, that is helpful. I will look in those code areas for possible
issues.

maudette01432, 2020-02-12:
Bart,  I disabled wifi/tcp/bt and all the wireless stuff.  The core serial stuff works fine (no surprise GRBL core has been solid for a while now).  It's only when wifi is enabled that I'm getting crashes.  I'm guessing something isn't feeding the WDT or if something establishes an additional connection while it's running it crashes.  I'm ran the  GCODE a couple more times just to see if it was a fluke - but this is the first time it got through an entire GCODE w/o the WDT reset (serial only).  So this might help to narrow the search a little bit.

maudette01432, 2020-02-12:
Just FYI - I did a clean linux install on a different machine and recompiled and still get the random "connection error" reboot (no serial/no BT).  I just wanted to make sure there wasn't something sullied on my system.

bdring, 2020-02-13:
Was working on this issue and found a big problem. [See this issue](https://github.com/bdring/Grbl_Esp32/issues/329)

I don't know if it is related.

maudette01432, 2020-02-19:
Bart,  I'm guessing it's related.  Hoping a solution is found.  There are several other open issues that appear to have similar symptoms as this one.

maudette01432, 2020-06-22:
Any word on the serial issues?  My ESP32GRBL board is more-or-less an anchor without a working compile for the issue above...or could someone provide me a working compile if I send in a config)?

bdring, 2020-06-22:
If you can send your machine definition and config.h, I can compile for you.

MitchBradley, 2020-06-22:
I would consider soldering a bulk decoupling capacitor - say 100 uF - from 5V to GND near the ESP32 module, if there is not already one.  WiFi can take a lot of current sometimes.  Not sure it will help but it is an easy thing to try.