bdring/Grbl_Esp32 #329: New Serial Port Problem

bdring, 2020-02-13:
[This commit](https://github.com/espressif/arduino-esp32/commit/ed220bd0420236e9ff95888f7459f0eb9c97a570) appears to have broken the USB/Serial port streaming in Grbl_ESP32.

I get random, but regular errors when streaming gcode. If I use an espressif/adruino-esp32 commit before that, everything works fine. 

You can change to a commit the day before this with 
"git checkout b50a1755c8b027826d6adce3ca189fc3ea644e90"

bdring, 2020-02-13:
[This pull request is looking promising](https://github.com/espressif/arduino-esp32/pull/3713).  Still testing.....

maudette01432, 2020-02-26:
Bart,  what is the process to grab this pull request into what I have and to re-compile and do some more testing.

bdring, 2020-02-26:
Just pull in the pull request or use the commit ID from before the problem (see first post)

Search github for help on doing this if you do not know how.

If you did not install using some form of Git, I don't know how to do it.

maudette01432, 2020-03-02:
Bart,  when I try to "git checkout" as above in the project it fails:

ma-lt6304:Grbl_Esp32 maudette01432$ !git checkout
git checkout b50a1755c8b027826d6adce3ca189fc3ea644e90 checkout
fatal: reference is not a tree: b50a1755c8b027826d6adce3ca189fc3ea644e90

I just cloned this repo this morning and I have used the "prefered" method via the boards manager to setup my boards.  Do I need to checkout/setup and build using the non-prefered method to get this branch of hardware libraries?

bdring, 2020-03-02:
I would use the second method using Git. I have not used the Arduino IDE boards manager method.

bdring, 2020-05-05:
Changes merged woot!

https://github.com/espressif/arduino-esp32/pull/3713

closed 2020-05-05 by bdring