bdring/Grbl_Esp32 #330: Encoder support

arcadeperfect, 2020-02-14:
Any thoughts on closed loop encoder support?

BarbourSmith, 2020-02-14:
I second this!

bdring, 2020-02-15:
Are you actually asking for closed loop motor support? I think that is best left on board.

BarbourSmith, 2020-02-16:
I am planning to work on this in a fork starting pretty soon. Maybe it's the kind of thing that we can take a look at in my fork once it's working and discuss if it has a place in the main branch then. 

What I'm imagining is having a module which replaces the step pulse generation with a closed loop control generation...everything above that would remain the same.

bdring, 2020-02-16:
Would it be for brushed DC motors or 3 phase brushless DC motors? IMHO I think the brushless is the way to go for power density and cost. 

Are you looking at ABN type encoders or magnetic absolute encoders?

One thing I was considering looking at is the built in encoder support on the Trinamic 5160 drivers. I cannot close the loop, but it would allow monitoring the loop without adding any extra ESP32 I/O pins.

BarbourSmith, 2020-02-17:
I agree that the three phase brushless DC motors are the way to go, but I need reduction gears after motor (something like 300:1) and because brushed DC motors are often gear reduced for things like paper shredders and those little barbie cars I can get them a lot cheaper than I could anything brushless...at least that I have seen.

I'm planning to use the AMS magnetic encoders because I need pretty high resolution and for the cost they seem like the best (although the 2% non-linearity is not ideal and will require some calibration). They talk over SPI so they don't end up taking up too many pins.

bdring, 2020-02-17:
The robot dogs, like the MIT Cheetah use brushless DC motors. They are typically only geared down 6:1 and they can do back flips.

I have one of those motors and it is a beast. I actually think it is more powerful than needed for CNC. I would rather go with about 3:1 and take advantage of the higher output RPM.

I also have an ODrive brushless setup.

BarbourSmith, 2020-02-17:
That's a fantastic point. I can easily do a 10:1 reduction with a belt.

The cost is still a factor though. The brushless drivers are a lot more expensive.  

I'm going to look into this more.

dedekcz, 2020-02-17:
I also vote for some encoder support. 
I have several DC brush motors (Faulhaber 3863) + ABN type encoder  (HEDS-5500) and my plan is use them for OpenPnP machine. I actually want to use Tarroco (DRV8701 + CY8C4245)  board to control this setup. But any level integration with ESP looks interesting.

and of course, many thanks @bdring  for outstanding open source projects!

BarbourSmith, 2020-02-17:
Since there are so many possible combinations (brushless vs brushed, ABN vs SPI) it seems like this feature request could pretty quickly grow to be either too big or too limited. Is it possible to make the stepper driver code somewhat modular so that an alternative module could be swapped in to control different motor types? 

Something like 
UI -> Gcode -> Motion and Acceleration Planning -> Step generation 
vs 
UI -> Gcode -> Motion and Acceleration Planning -> Closed loop signal generation

That would allow the core code to be reused for multiple applications and at the same time not rock the boat too much in terms of changing what is already fantastic

MitchBradley, 2020-02-17:
stepper.cpp is already ifdef'ed out the wazoo for different scenarios.  The Arduino IDE doesn't really lend itself to modular design.  It was intended for simple novice hobbyist projects and when things get complicated, the code starts to look messy. But maybe there is some hope.

I think the bottom line with respect to closed loop encoder support is "what is the value of this compared to the many other improvements that could be made, and how should the available development resources be allocated".  My personal opinion is that, while encoder support does indeed have value, it is not the highest thing on the list.

bdring, 2020-02-17:
I agree, the #ifdef'iness is a pain. 

If you look at [custom_machine_template.cpp](https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/custom_machine_template.cpp), it is a little bit of a cleaner way of adding user defined features.

With that said...Internally closed loop motors are not driven by step and direction. They are chasing a target. That target is the current Grbl position. That value is globally available. You are going to to have to create a new task or interrupt. It could just look up the target itself.

I have a decent amount of experience with building closed loop machines and the biggest take away is they are great, but a wiring mess. The Cheetah motor was a much different experience. You supply power and CAN. Both of those can be daisy chained. I personally think the future of DIY closed loop is in fully integrated brussless DC motor/drivers. You just send power and CAN or Step /Dir.

derFrickler, 2020-02-17:
If you want something integrated as Nema17/23/34 take a look at the JMC CL-Steppers and Servos, they are great and reasonable cheap.

MitchBradley, 2020-02-17:
I agree.  Closed loop steppers are quite affordable now, and work very well.  I use them on X and Y for my Taig mill, with excellent results.  I use the integrated variety, so the tuning is done at the factory, there are no high-current pulses running around to cause EMI, and the system layout is compact, with no need for a box to hold the drivers.

MitchBradley, 2020-02-17:
At a slightly lower price point, there are MKS SERVO42B (NEMA 17) and MKS SERVO57B (NEMA 23).

arcadeperfect, 2020-02-17:
Forgive my ignorance... what's the idea behind a closed loop stepper? Do you still control it with direction / step pulses? Or do you send it a target value like a servo?

BarbourSmith, 2020-02-17:
> That target is the current Grbl position. That value is globally available. You are going to to have to create a new task or interrupt. It could just look up the target itself.

Fantastic! That sounds like a lot of the architecture is already in place. Would anyone be willing to give me a two sentence road map for where you would start if your were going to give it a go?

How does one grab the GRBL position globally? If I were to add a new timer which one should I use without stepping (pun intended) on existing code?

MitchBradley, 2020-02-17:
A closed loop stepper is a stepper and encoder and driver and encoder sensing electronics all rolled into one.  You send it step and direction, and it figures out what to do. It won't lose steps; if it starts to get behind, it increases the motor current as necessary, and if it just can't do it, it goes into a fault mode and stops.  I think they are the wave of the future.  There are also versions that can receive target positions via CAN bus etc, but the step/dir ones are cheaper and out-of-the-box compatible with GRBL and friends.

bdring, 2020-02-17:
I have some closed loop steppers and they work as advertised, but they have less torque, and a lot lower max speed than similar sized brushless DC motors. I tried to retrofit and Chinese 3020 mill with them and while they were closed loop, the performance was not as good as the original steppers.

karoria, 2020-02-18:
I am using leadshine easy servo (closed loop steppers) with 85kgcm torque rating in all my 3 axis; x, y and z. I am having excellent results. I am using grbl_esp32 with cncjs with usb serial. My minimum move (resolution) is 0.002 mm and at this resolution I can have top speed of 8000 mm/min without any dimensional glitch. I am trying to upload photo/video of one of the jobs here.

bdring, 2020-02-18:
One nice feature that some closed loop system have is the ability to turn off the torque, move the motor manually and have to controller update its position,

I have that here with this Grbl_ESP32 delta. When the motion stops, you can manually move the end effector. Grbl_ESP32 knows where you moved it to, so the next move is very smooth and controlled.

https://www.youtube.com/watch?v=wPDP6ckP0sQ

It is a great feature for a mill. Often you want to use it in a manual mode. It is really nice to have DROs update when manually moving it.

It can also be very important for homing on some machines with exotic kinematics. There may be certain positions where it is important to home correctly or you could accidentally stress joints.

Nairod785, 2020-02-25:
Take a look at some of these integrated encoder servo motor reaching up to 400W
https://www.aliexpress.com/af/servo-motor-integrated.html

Sorry I'm still learning about this project, would the board be currently compatible with such servo motors? Thx :)

bdring, 2020-02-25:
Those drivers tend to take step and direction signals. Grbl_ESP32 can drive that fine, but they tend to be 5V and opto isolated. That means you need a 5V signal that can drive a few milliamps. The ESP32 is 3.3V and many basic voltage level shifters can't drive any current.

I will be selling this controller in a few weeks. It is designed to run those drivers. Basically it uses a 74ACT245 chip to drive the optos.

[image]

Nairod785, 2020-02-25:
Thank you for the fast reply! 
Woow this board would suit well these servos then
Just to make sure, would it still be compatible with a CO2 laser control? I think it requires just a digital PWN/TTL Pin right?

Thank you for the work on this project!
I will most certainly buy one of your board on tindie and use it for my big CNC Laser Cutter
Seems like the ESP32 implementation to the GRBL world is huge.

MitchBradley, 2020-02-25:
There are PWM and EN pins on the connector at the lower left that will suffice for laser control.

svalambhia, 2020-04-09:
Here are couple of good examples of converting stepper to servo or brushed dc 775 to servo mostly with code and schematics

**Stepper to servo converter**
https://github.com/jcchurch13/Mechaduino-Hardware
https://github.com/makerbase-mks/MKS-SERVO42B
https://www.crowdsupply.com/citrus-cnc/tarocco#details-top

**775 Motor**
https://github.com/misan/dcservo & https://www.youtube.com/watch?reload=9&v=kTUaKS-4Spc

ZuurtjeNL, 2020-06-18:
> I will be selling this controller in a few weeks. It is designed to run those drivers. Basically it uses a 74ACT245 chip to drive the optos.

Any update for this board? Where to buy?

BarbourSmith, 2020-07-16:
I am finally getting around to giving this a proper crack and I was wondering if the motor classes described in #449 might be relevant and the best way to do this. All I really need is the current machine coordinates (where it should be), and a place I can call my PID function regularly.

Given all the changes and improvements since this issue was opened what are your thoughts on the best way to get started?

MarcinanBarbarzynca, 2020-08-10:
https://easyeda.com/polaski/esp32-cnc-shield

karoria, 2020-08-11:
> Those drivers tend to take step and direction signals. Grbl_ESP32 can drive that fine, but they tend to be 5V and opto isolated. That means you need a 5V signal that can drive a few milliamps. The [...]

Hi @bdring, I can't see this borad on tindie. Have you started selling this? I am interested to buy.