bdring/Grbl_Esp32 #331: Problems Compiling Firmware

MitchBradley, 2020-02-15:
Please answer the following questions:

Have you read the [wiki regarding how to compile](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)?
Yes

What version of the Arduino IDE are you using? 
1.8.12 - just downloaded it today

What version (commit date) of the [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) are you using? 
1.0.4 

Are you using the master branch of Grbl_ESP32?
Yes, cloned yesterday.  Commit is 2451d37

Have you made any edits or configuration changes (list them) to the  firmware?
No

The compilation succeeded - my issue is with the instructions in "Compiling-the-firmware"

1.  The board settings dialog in the latest Arduino IDE differs from that shown in the instructions.  The "WiFi101 Firmware Updater" choice is now listed as "Wifi101/WiFiNINA", and the closest choice to "Minimal SPIFFS (Large APPS with OTA)" is now "Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)"  (Obviously I am being picky here, but the instructions are quite precise and picky themselves - a good thing - so I feel justified in being picky too :-)
2. The instructions assume that you know how to open an Arduino project.  It should be spelled out - File>Open... , browse to the directory containing Grbl_Esp32, open Grbl_Esp32.ino.

bdring, 2020-02-15:
Thanks...updated

closed 2020-02-15 by bdring

karoria, 2020-02-15:
@MitchBradley 
I am glad to see you here. Meet Mr. Barton @bdring who is a firmware genius. And Mr. Barton, I would like to introduce Mr. Mitch as a javascript genius and application expert who is chief contributor of cncjs and g2core.

MitchBradley, 2020-02-15:
Thanks for the intro, but I would like to set the record straight.  I am barely competent - and maybe not even competent - at Javascript.  I fought my way through the CNCjs code to mostly figure it out but every step was a struggle. I would not hire myself for a job that was primarily focused on JavaScript development.

I will, however, admit to having some firmware chops. I was the architect and author of Open Firmware, the boot firmware that, during its heyday, was used on pretty much all RISC servers, from Sun Microsystems, IBM, Apple, HP, etc.  I was also the firmware lead at One Laptop Per Child.  Bare metal is my forte; top of stack is a place that one sometimes goes by necessity.

Now that I am retired, I make golf putters from bamboo using a variety of CNC machines, and invent gadgets to help local farmers and food producers.

bdring, 2020-02-15:
Same here. I consider myself a machine designer and firmware is just a necessary evil. 

...also retired (from real jobs)

karoria, 2020-02-15:
OMG, You are much more than what I already known so far... Keep your good work going for the betterment of open source community. By the way, grbl_esp32 is a great project lead by @bring very proactively. Best part of it is it doesn't require host PC, though I use it with cncjs because of its rich interface.  Still, RTOS is a wonderful thing and it doesn't require formal shutdown. The web interface is decent but requires attention for being feature rich like cncjs. I am already using this firmware in my vmc and getting excellent results with a rigourously tested 120kHz pulse for each axis and realtime overrides. This is for your quick go through.

karoria, 2020-02-15:
> Same here. I consider myself a machine designer and firmware is just a necessary evil.
> 
> ...also retired (from real jobs)

That is your greatness... A real talent is always wrapped in politeness.