bdring/Grbl_Esp32 #333: Hard buttons for all essential functions

karoria, 2020-02-17:
**Please describe the feature you would like implemented**
I would like a way to control my machine with button switches instead of touch for ease of operating and less chance of misfiring any funtion accidently. Think of the dedicated pins for Hold, Start, Reset and Door currently present in the firmware and work extremely reliably. The same way, I request dedicated pins (I know we are running short of pins but there might be a solution like adding one more esp32 board hard wired with the first one). What I request is following functionality at least:
1. toggle mist/flood coolant on/off
2. toggle spindle on/off
3. feed override -10%
4. feed override +10%
5. feed override reset (to 100%)
        (An incremental encoder can also be furnished for overrides which give excellent user friendliness. Of course with a little different coding. Middle push button can be allocated to reset override to 100%)
6. spindle override -10%
7. spindle override +10%
8. spindle override reset (to 100%)
9. Rapid override 25%
10. Rapid override 50%
11. Rapid override 100%
12. jog cancel (can be combined with feedhold to save a pin)
13. kill alarm lock (unlock)
14. home
15. probe (cycle can be defined in firmware only)
16. jog distance selection (I suggest 4 dedicated pins like 0.01, 0.1, 1, 10 in mm, however it can be user defined in firmware)
17. X+, X-, Y+, Y-, Z+ and Z- (6 dedicated pins)

TOTAL 25 PINS ARE REQUIRED FOR ALL ABOVE WORK. CAN BE MORE OR LESS AS PER UNIVERSAL SUITABILITY. IN CASE OF BUSY SCHEDULE OF DEVELOPERS, YOU CAN GUIDE US HOW TO DO THOSE THINGS IN FIRMWARE BY GIVING AN EXAMPLE.

**Why do you think this would improve Grbl_ESP32?**
grbl_esp32 is excellent standalone firmware and above dediacated pins will make it more reliable in terms of operating.

**What do you need the feature for?**
Ease of operating and "No use of touch" for a standalone system will be more versatile.

**Will this feature appear to a lot of users?**
Yes.

derFrickler, 2020-02-17:
Nice Idea, but i would go and use a pendant with its own controller for that. 

Many of that stuff can be done via realtime commands on the serial interface. 
You can use a 2nd controller (Arduino with enough pins) and just make it sent these commands via the serial interface of grbl (if you don't use the USB-Serial connection)
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands

Check #214 as it might do a part of that. 

Maybe as an option it would be nice to have a 2nd serial RX/TX interface in Grbl_ESP32 to use for this, so that one can still use the USB with a normal PC Client.

karoria, 2020-02-17:
@derFrickler thanks for quick response. I am a user only and not a software/firmware expert. But I can do it if you guide me properly. I have also tried with pro micro and it runs ok but it requires a host pc as it sends shortcut keys as hid keyboard. Further, arduino keyboard.h library doesn't support extended ascii codes to be sent.
I want to know more about using another esp32 board connected via tx/rx as my usb serial is busy and how to send those ascii characters or other suitable commands.

closed 2020-03-30 by bdring