bdring/Grbl_Esp32 #337: Zero Cross Detect for AC Dimming Board Spindle Control

reynolds087, 2020-02-19:
**Please describe the feature you would like implemented**

Would you consider adding support for AC PWM controllers with zero-cross detection. Example: https://www.amazon.com/gp/product/B072K9P7KH/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1

I tested it with and esp8266 and code from this repository: https://github.com/nassir-malik/IOT-AC-Light-Dimmer-With-Alexa.git to control a 110VDC brushed DC motor and it works  (full wave bridge rectifier after the output from the controller to convert AC to DC)

**Why do you think this would improve Grbl_ESP32?**

I haven't been able to find affordable PWM controllers that support 110VDC. The cost of the AC dimmer and a bridge rectifier is approx $10, and it is rated up to 3kW.

**What do you need the feature for?**

I need the feature so that I can simplify my setup. Currently using GRBL with an arduino uno, and using the PWM from spindle output (low pass filtered with R/C circuit and voltage divider to 3.3v) to the ADC on the esp8266. Then driving the AC dimmer/Rectifier circuit with output from the esp8266. All of this could be handled locally on an esp32. It would just require an extra pin for the zero cross detection.

**Will this feature appear to a lot of users?**

I think it would because the 500 and 800 watt 110VDC spindles are pretty cheap and readily available, and it's hard to find an affordable GRBL-compatible power supply for them.

bdring, 2020-02-26:
I am not interested in support this type of speed control. Most routers have a built in speed control for this. Additionally, it works by reducing power to the router. This means the router is weak at slower "speeds" and has difficulty maintaining that lower speed.

reynolds087, 2020-02-26:
This is not for a router. It's for a DC spindle motor. Here is an example:

https://www.ebay.com/itm/500W-110V-DC-Spindle-motor-52mm-diameter-DC110V-12000rpm-high-torque-air-cooled/222596352678?hash=item33d3c6caa6:g:owoAAOSws2RcQskw

The power supplies that come with them are controlled by a potentiometer. The ones that support pwm are more expensive. Additionally, I don't think the controller reduces power to the motor, only voltage. DC controllers for brushed motors all work this way, don't they?

bdring, 2020-02-26:
I really don't think that power supply is supposed to be dimmed.

What does the pot do? If is outputs a voltage to the power supply, there are easier ways to control it.

for starters....Send a close up picture of the pot connections.

MitchBradley, 2020-02-27:
Many such motor controllers that are controlled by a potentiometer use a 555 timer IC to switch a MOSFET on and off, so in effect they are doing PWM internally.  The potentiometer, which is acting as a variable resistor, controls the charging rate of a capacitor.  In that type of circuit, you would not be able to send a voltage in - but you could replace the 555 with a direct PWM output.

The notion of "only reducing voltage, but not power" is not a useful way of thinking about it. If you you reduce voltage to a mostly-constant resistance device like a motor, the power is reduced according.  Power is voltage squared divided by resistance, so when one goes down the other goes down.

PWM controls motor speed by reducing the average voltage by turning the voltage on and off many times per second,   That is "efficient", by which I mean "wastes little energy as heat and can be done with inexpensive components".  PWM is efficient is because the switching transistor is either fully on - very low resistance - or fully off - very high resistance - for the vast majority of the time.  Only when it is in the brief transition from on to off is it in an intermediate state where significant power can be lost by heating of the transistor.

"Brushless DC" motors are actually AC motors with circuity that takes a DC input and creates a mult-phase AC signal to drive the motor.  That AC signal is often created using PWM techniques, for the same efficiency reasons, but there is extra complexity around switching the polarity for the various motor phases and doing so at the desired speed.

That triac dimmer looks like a dangerous accident waiting to happen, unless one is skilled in line voltage wiring - not in the skill set of the average GRBL user.

bdring, 2020-02-27:
@MitchBradley The pot has three terminals, so it looked like a voltage divider to me. 

I have used a few of those DC spindles in the past. They are often designed for use in cheap Chinese CNC routers. The ones I have used will also accept a PWM signal instead of a pot voltage. They might have a simple low pass filter on the input. They usually give a recommended frequency around 5kHz.

Tip: AliExpress is cheap, but Amazon often sells the same thing. There is often more detailed information and helpful discussion in the reviews and questions section.

reynolds087, 2020-02-27:
I think perhaps I have explained my hardware configuration poorly. I am NOT using one of the power supply that comes with those 52mm spindles. Those power supplies only accept controlling by 0-10 volts analog or you can use the external pot. I am controlling the spindle automatically with GRBL and PWM. You mentioned that safety is the main concern, and I think you are right because you have to be careful with the line level voltage. 

Safety concerns aside, I will say that it works very well.   I use a variation of the code from the repository I linked to, running on an ESP8266 to control the voltage output from the triac dimmer/rectifier combo. So the output to the spindle is controlled by Candle (GRBL Controller) from  0 - 110 VDC.

If using GRBL on an ESP32, Is there a way for me to reference that library myself and use my spindle control method rather than having to use the additional ESP8266 to accomplish this?

MitchBradley, 2020-02-27:
It is certainly possible, but you are asking people to expend a fair amount of their precious time so you can eliminate a $2 component.

reynolds087, 2020-02-28:
Fair point. It just seemed like a nice all in one solution, but I can use my workaround.

karoria, 2020-02-28:
@reynolds087 
I would rather recommend a 3.3V PWM to 0-10V analog converter which is easily available in market so that you can use the same driver which came with your spindle so that you don't have to mess with output torque at low rpm. See this link:
https://www.ebay.com/itm/3-3V-PWM-100HZ-3KHZ-Signal-to-0-10V-Voltage-Converter-D-A-Digital-Analog-Module-/163596299408

reynolds087, 2020-02-28:
@karoria Can you explain why you think I am getting low torque at low RPMs? I haven't noticed that issue. Wouldn't you assume that the power supply that it comes with is also using a triac and a rectifier circuit to convert from AC to DC. Both systems vary from 0-110 volts DC on the output side to change the rpm of the spindle.

Is there some fundamental difference in the architecture of the two power supplies that I am missing?

reynolds087, 2020-02-28:
I am only really familiar with two commonly used topologies for AC/DC power supplies, but I'm sure there are others. I assumed it was  either a switch mode PSU or just the combination of a triac rectifier with filtering and smoothing to mitigate the effects or ripple. I have never seen a 110VDC switch mode power supply, at least not an affordable one; there may be one out there. That is why I am assuming they are using the triac topology. I didn't buy one of the included PSUs because it's like an additional 40 bucks, and doesn't support PWM. My method costs 10 dollars.

MitchBradley, 2020-02-28:
As an electrical engineer, I can explain it.  The way a triac dimmer works is as follows. At the beginning of an AC half cycle, the control circuit either turns on the triac or not.   The triac then stays on until the end of the half cycle when the current drops to zero - which is not necessarily exactly aligned with the zero voltage point because inductance in the load circuit can cause phase shift between the voltage and current waveforms. It is not possible to turn off a triac via the control input.  One they are on, they stay one until the load current drops to zero - which is why they are used in AC circuits.

What this means is that your control is limited to half-cycle granularity - on a 60 Hz system, you get to turn on in units of 16 milliseconds.  If you want to run at 1% speed, you supply full power for 16 milliseconds and no power for 1.584 seconds. The motor can't average that out; it bumps along instead of turning smoothly. The motor is getting only 1% of the total power, averaged over time, that it would receive if the triac were turned on on every half cycle.  The motor has to start and stop on every bump, so the inertia of the spindle is coming into play over and over.
On a PWM system, the power to the motor can be switched on and off at a higher frequency - you are not limited to half-cycles of the AC line frequency.  The switching element is typically a MOSFET, which can be turned both on and off, so you can generate a pulse train at much higher frequencies than a triac.  This has the advantage that the inductance of the motor can smooth out the current waveform.  The voltage waveform is still bumpy, but the bumps are much closer together so the motion is smoother.  The frequency can be adjusted to work best for the motor.

You are still getting only a fraction of the average power at low speeds, thus the low end torque is limited, but at least the motion is not jerky and you do not have to deal with the extra loss from inertia.

Lower torque at low speeds is pretty much inherent in DC motors. To get good torque at low speeds you need to use a different kind of motor, such as a brushless DC setup, which is really an AC motor driven from special electronics that synthesizes an AC wavefrom from a DC input voltage.
If your setup is giving you acceptable results at the price point that you like, so be it, but it inherently involves a lot of compromises. It is using the components in ways they weren't really designed for.  Sometimes you can get away with that and sometimes you can't.

You get what you pay for.

I am closing this issue because it has been beaten to death and it is unlikely that any new information is likely to surface.

closed 2020-02-28 by MitchBradley