bdring/Grbl_Esp32 #343: Assert fail error while job is running.

povi2946, 2020-02-28:
It kept crashing while doing jobs from SD card so I just used UGS to send gcode via USB.

For most simple jobs it does ok, but it always crashes on raster jobs. Gcode is generated by LaserWeb4.

>>> G1X98.82Y127.20S255.00F750
Skipping blank line #723
Skipping comment-only line: (Pass 0 Path 177)
ok
>>> G0X98.74Y127.40
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1446 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x4009138f on core 1
Backtrace: 0x40095054:0x3ffbe880 0x40095285:0x3ffbe8a0 0x4009138f:0x3ffbe8c0 0x400f907b:0x3ffbe900 0x400e0361:0x3ffbe920 0x400e047d:0x3ffbe940 0x4008147d:0x3ffbe960 0x40087685:0x3ffbe980 0x4008301d:0x3ffba170 0x40089ed6:0x3ffba190 0x40085a7f:0x3ffba1b0 0x400916c1:0x3ffba1d0
Rebooting...
[MSG:TMCStepper Init using Library Ver 0x000601]
[MSG:X Trinamic driver test passed]`

bdring, 2020-02-28:
Please provide all of the startup messages, plus your $$ settings and attach the gcode file. I want to repeat your setup and try the code.

povi2946, 2020-03-01:
Thanks, I am using Pen Plotter TMC2130 v1. I will be able to attach settings and g-code only on Monday. 

Adding config and cpu_map . maybe they would help too. 
[config_and_cpu_map.zip](https://github.com/bdring/Grbl_Esp32/files/4272437/config_and_cpu_map.zip)

povi2946, 2020-03-02:
[gcode_and_startup.zip](https://github.com/bdring/Grbl_Esp32/files/4276389/gcode_and_startup.zip)

Here attaching my startup output which includes all $$ settings. and Gcode that I was running. Also another piece of output when I received the error.

Thank you

bdring, 2020-03-02:
I checked out what you sent. I noticed you are not using the latest code. 

A commit 27 days ago added a message that shows the what SDK version of the ESP32 core was used. See my messages below.

You are also using laser mode which has very heavy use of the PWM channel. That was recently optimized.

I loaded your settings and ran the code. The only difference was my controller version. That should have no impact. I was able to run the complete job from both UGS (steaming)  and via the WebUI (SD card) and both ran fine. I ran it twice from each place.

I suggest you update to the latest master. also [see this note issue about the core version](https://github.com/bdring/Grbl_Esp32/issues/329) 

[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using cpu_map:ESP32_TMC2130_PEN V2]
[MSG:TMCStepper Init using Library Ver 0x000602]
failed.[MSG:X Trinamic driver test failed]
[MSG:X Trinamic driver test check motor power]
failed.[MSG:Y Trinamic driver test failed]
[MSG:Y Trinamic driver test check motor power]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Client Started]
[MSG:Connecting Barts-WLAN]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connected with 192.168.1.12]
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
Grbl 1.1f ['$' for help]

povi2946, 2020-03-02:
I pulled the newest, but I still get the error. I wasn't running Gcode via USB. I started the the job from webUI but I had serial connected for debugging. This time it was in different half way through it just stopped moving Y axis it was only X. I downloaded some jpegs from laserweb to etch for calibration purposes all those files and rasters went fine.. It moves and homes with no problem. But whenever I wanted to do a proper job it gets errorous.

povi2946, 2020-03-02:
[image]
You can see in the kinda OK etch was raster just doing horizontal. But then at the top a non-sense was raster diagonal. It seems that in all diagonal jobs it gets glitchy, also on the side is visible when it just got stuck at one horizontal like and Y stopped moving. In general machine moves freely. I still need to calibrate more but I get these problems.

bdring, 2020-03-03:
I ran my WebUI jobs the same way. I had a Serial GCode sender open and constantly requesting status, but the SD card job was started and monitored from the WebUI.

I do not have a full laser machine I can test with. I put an LED on the spindle output to visually monitor the the output.

I don't know why it would stop moving in one axis while still running a job. My first guess would be an electrical or thermal issue.

povi2946, 2020-03-03:
I see. Thank you. I am running two Y-motors on a single driver. They were not getting hot, but is it possible that driver TMC2130 shuts them off?
We have these extended settings 
$140 thru $145 – [X,Y,Z,A,B,C] Stepper Run Current (extended settings)
This is for Trinamic SPI stepper drivers only
$150 thru $155 – [X,Y,Z,A,B,C] Stepper Idle Current (extended settings)
Is there is a maximum value I can put in? 

also, I saw my TMC library is one build older, I will try to update that one too. And test again. Thank you.

bdring, 2020-03-03:
2 motors on one driver is unsupported territory.

The drivers try to read the motor during operation and could get confused by reading two motors.

I would contact Trinamic about it.

povi2946, 2020-03-03:
Ok, so this was most likely the problem for Y axis. 
I have this Step,dir,en splitter board.

[image]

I could use it split step direction signal to two Trinamic drivers. But then obviously my TMC2130 pen plotter board looses purpose in a way. I just want to ask for a recommendation for my next approach.  Should I try to hack pen plotter board to get step dir signal on the pins and then use the splitter board. Or just use Grbl_ESP32 Development Board 4.1v, and if I am correct I saw that in config we have options to use two drivers for the same axis. so I could use Z and Y  slots for two Y axis motors.

MitchBradley, 2020-03-03:
Either approach can be made to work.  On the splitter board, I don't see any way to get the SPI pins over to the pen plotter board, but you could solder on some wires, or you could use the TMC drivers in standalone mode, without SPI.  Search for "tmc2130 standalone mode" to learn how.
The Grbl_ESP32 Development Board solution will result in a cleaner setup, but since you already have the other ones it might be a faster way to get going.

bdring, 2020-03-03:
That is an interesting add-on board, but it has some issues. With SPI drivers you would need to hack in some external wires. I am not sure how it handles the micro-stepping jumpers with non-spi drivers.

The 3 axis controller can be ganged, but it does not support Trinamic drivers in SPI mode.

MitchBradley, 2020-03-03:
This video shows how to modify TMC2130 to use standalone mode, so you could use them with SPI
https://www.youtube.com/watch?v=Jh8iwqc1TP0

povi2946, 2020-03-03:
Thank you, yes I will use TMC2130 in a standalone mode and gang the axis. I will give another attempt tomorrow with Grbl_ESP32 Development Board. It should fix the Y problem for sure. I am still not sure how this could have affected the error I kept receiving but if tomorrow it's all good maybe we can close this issue. Thank you again

MitchBradley, 2020-03-03:
It is possible that, with 2 motors connected, the current feedback from the motors was causing the driver chips to enter a protection state.