bdring/Grbl_Esp32 #348: "defaults.h" overrides values defined in "cpu_map.h"

Stefano-Orsolini, 2020-03-14:
I'm using the pin map CPU_MAP_ESP32_ESC_SPINDLE with great success but I noticed that "defaults.h" overrides the values defined in "cpu_map.h" for:
DEFAULT_SPINDLE_OFF_VALUE
DEFAULT_SPINDLE_MIN_VALUE
DEFAULT_SPINDLE_MAX_VALUE

when commenting this definitions in "defaults.h" I successfully control the brushless motor thanks to the values defined in "cpu_maps.h". 

no clue why this is happening or whether other definitions are affected.

MitchBradley, 2020-03-15:
Which version are you using?  In the latest master, those definitions are guarded by ifndef like this:

```
#ifndef DEFAULT_SPINDLE_OFF_VALUE
	#define DEFAULT_SPINDLE_OFF_VALUE 0.0 // $34 Percent (extended set)
#endif
```

The ifndef guards are supposed to prevent defaults.h from overriding values that are already set in cpu_map.h.  The guards have been present since at least 4 months ago (commit 8f4e7c20).

Stefano-Orsolini, 2020-03-15:
I guess I must have edited the parameters and created the problem myself.
I confirm that everything works using latest master as you stated!

closed 2020-03-15 by Stefano-Orsolini