bdring/Grbl_Esp32 #350: Driver S109

erikk12, 2020-03-15:
Hello, I'm trying to use this drivers (s109) but they don't work properly and I'd like to know if anyone has tried them before.

Thank you very much and sorry!

MitchBradley, 2020-03-15:
In principle it should work.  What exactly did you do and what exactly happened? Just saying "don't work properly" is not nearly enough information for someone to help you.  Details, details, details, please.

derFrickler, 2020-03-16:
I remember some problem with the S109 on the 3.3V Processor Boards. I think the VDD pin on the drivers had to be removed as they output 5V on that pin instead of needing 3.3-5V - but not sure, please check before.

bdring, 2020-03-16:
I agree with @derFrickler It does not appear to be compatible with 3.3V logic.

closed 2020-03-16 by bdring

derFrickler, 2020-03-16:
No, just checked again, the VDD Pin is a 5V output on the S109 drivers, that interferes - or maybe kills - the ESP32. If that pin is cut off the S109 driver board, it should work. Had that running some time ago.

erikk12, 2020-03-17:
Sorry for the late reply, @derFrickler thank you very much the VDD was the problem.

BigYuanHead, 2020-08-31:
I use tb67s109 moduels and removed the VDD pin, but it still don`t work properly. The moduel has output, I can feel the power on the motor, however, the motor just stuck there, it can`t rotate.  I check my wire, I can sure it is correct, since I use TB6600 stepper drivers for X, Y axises and they work nice.
This is the moduels that I use.
[image]

SouzaKoin, 2021-04-23:
@BigYuanHead  you said you use the TB6600 stepper driver, would you mind sharing the wiring schematic ?? i'm planing to use them on a school project but i saw some comments saying that it is not 100% compatible. thanks