bdring/Grbl_Esp32 #351: WebUI Contribution

beak90, 2020-03-21:
Hi quick question here. It looks like the index.html.gz that is in this repo is not quite the same as the ESP3D-WEBUI repo (grbl_6axis branch). Is the latter branch the correct development repo?

I would like to make some contributions to the Grbl_Esp32 Web UI, and the minified html file is of course not suitable for that.

luc-github, 2020-03-21:
Devt version of webUI is branch 2.1, the grbl_6axis branch was a pre devt version and has been already merged to 2.1 long time ago, so now shjould be different for sure. I do not maintain specific version for GRBL.
I only do PR to GRBL_ESP32 when there are critical changes affecting GRBL_ESP32 or per demand and when it is well tested on GRBL_ESP32.

Any issue with current version ?

kylebme, 2020-03-22:
Ah ok I didn't realize the 2.1 branch included the GRBL stuff. I see now that it auto detects the firmware and displays the correct stuff.

I'm interested in getting a 3d display in there. I'm going to try putting this into a panel: https://github.com/jherrm/gcode-viewer

I realize this will probably make the file too big, so I'm going to go with 2MB app, 2MB spiffs. I don't really care about OTA update. But maybe the gulp script could include a CLI parameter for whether or not you want the 3d viewer.

closed 2020-03-22 by kylebme