bdring/Grbl_Esp32 #353: Servo - how to set it up

dronecz, 2020-03-21:
Hi, 

I´m playing with servos (normal and continuous ) and I have bit problem with all what I tested. All off them start to move just after I disable alarm so I can move my axis. I followed information in servo part of the wiki, but there is not enough information to solve my issue. Does anybody know how to solve this? 

Thanks for your help.

bdring, 2020-03-21:
Continuous servos are not supported. Please describe exactly what you want the servos to do and we can help you.

Which axes use servos?
What range do you want to map the servo to?
What do you want it to do when the machine is homed?
how should it behave during an alarm?
How should it behave when the steppers are disabled?

dronecz, 2020-03-21:
Yes, I thought that, but I was hoping that it will act as normal servo but without min and max position. Maybe I´m wrong, but if can not detect exact position, it can be spin in one direction endlessly. One limitation can be defined map value, but if I will make this value big, I could work in large "area" and it could work. I was hoping that I will test that, but my normal servo behave same as continuous.

>  Which axes use servos?

I want use them for A, B and C asix

> What range do you want to map the servo to?

I do not know for now. I want to test some values and see what is possible. 

> What do you want it to do when the machine is homed?

It should do nothing. It should work when other axis are at work. 

> how should it behave during an alarm?

It should stop. 

> How should it behave when the steppers are disabled?

It should be disabled as well.

bdring, 2020-03-22:
What controller are you using and what hobby servo type?

dronecz, 2020-03-22:
If you mean what board I´m using than I designed my own board (picture at showcase channel in Slack). U use 5g (HXT500) and 9g (HXT900) servos. Here is my board definiton: 

```
#define CPU_MAP_NAME "CPU_MAP_SHAPE_LIKE_MEGA"

// Be sure to change to N_AXIS 6 in nuts_bolts.h
#ifdef N_AXIS
#undef N_AXIS
#endif
#define N_AXIS 5

// stepper motors
#define USE_RMT_STEPS

#define X_STEP_PIN        GPIO_NUM_12
#define X_DIRECTION_PIN   GPIO_NUM_26

#define Y_STEP_PIN        GPIO_NUM_2
#define Y_DIRECTION_PIN   GPIO_NUM_4

#define A_STEP_PIN        GPIO_NUM_27
#define A_DIRECTION_PIN   GPIO_NUM_33

#define Z_STEP_PIN          GPIO_NUM_14
#define Z_DIRECTION_PIN     GPIO_NUM_25

//#define B_STEP_PIN        GPIO_NUM_15
//#define B_DIRECTION_PIN   GPIO_NUM_32

// C is a servo

// servos
    #define USE_SERVO_AXES
//    #define SERVO_Z_PIN       GPIO_NUM_22
//    #define SERVO_Z_CHANNEL_NUM   6
//    #define SERVO_Z_RANGE_MIN   0.0
//    #define SERVO_Z_RANGE_MAX   10.0
//    #define SERVO_Z_HOMING_TYPE SERVO_HOMING_TARGET // during homing it will instantly move to a target value
//    #define SERVO_Z_HOME_POS    SERVO_Z_RANGE_MAX // move to max during homing
//    #define SERVO_Z_MPOS        false   // will not use mpos, uses work coordinates

#define SERVO_B_PIN       GPIO_NUM_15
#define SERVO_B_CHANNEL_NUM   4
#define SERVO_B_RANGE_MIN   0.0
#define SERVO_B_RANGE_MAX   10.0
//#define SERVO_B_HOMING_TYPE SERVO_HOMING_TARGET // during homing it will instantly move to a target value
//#define SERVO_B_HOME_POS    SERVO_B_RANGE_MAX // move to max during homing
//#define SERVO_B_MPOS        false   // will not use mpos, uses work coordinates   

// limit switches
#define X_LIMIT_PIN       GPIO_NUM_34
#define Y_LIMIT_PIN       GPIO_NUM_35
#define Z_LIMIT_PIN       GPIO_NUM_36
#define A_LIMIT_PIN       GPIO_NUM_39
//#define B_LIMIT_PIN       GPIO_NUM_   
#define LIMIT_MASK        B1111

// OK to comment out to use pin for other features
#define STEPPERS_DISABLE_PIN GPIO_NUM_13

#ifdef HOMING_CYCLE_0 // undefine from config.h
#undef HOMING_CYCLE_0
#endif
//#define HOMING_CYCLE_0 (1<<X_AXIS)
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))
//#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS) |(1<<A_AXIS)|(1<<B_AXIS))

#ifdef HOMING_CYCLE_1 // undefine from config.h
#undef HOMING_CYCLE_1
#endif
//#define HOMING_CYCLE_1 (1<<A_AXIS)
#define HOMING_CYCLE_1 ((1<<A_AXIS)|(1<<B_AXIS))

#ifdef HOMING_CYCLE_2 // undefine from config.h
#undef HOMING_CYCLE_2
#endif

  #define HOMING_CYCLE_2 (1<<Y_AXIS)

  #ifdef HOMING_CYCLE_3 // undefine from config.h
  #undef HOMING_CYCLE_3
  #endif
  #define HOMING_CYCLE_3 (1<<B_AXIS)

#define DEFAULT_STEP_PULSE_MICROSECONDS 3
#define DEFAULT_STEPPER_IDLE_LOCK_TIME 200 // 200ms

#define DEFAULT_STEPPING_INVERT_MASK 0 // uint8_t
#define DEFAULT_DIRECTION_INVERT_MASK 2 // uint8_t
#define DEFAULT_INVERT_ST_ENABLE 0 // boolean
#define DEFAULT_INVERT_LIMIT_PINS 1 // boolean
#define DEFAULT_INVERT_PROBE_PIN 0 // boolean 

#define DEFAULT_STATUS_REPORT_MASK 1

#define DEFAULT_JUNCTION_DEVIATION 0.01 // mm
#define DEFAULT_ARC_TOLERANCE 0.002 // mm
#define DEFAULT_REPORT_INCHES 0 // false

#define DEFAULT_SOFT_LIMIT_ENABLE 0 // false
#define DEFAULT_HARD_LIMIT_ENABLE 0  // false

#define DEFAULT_HOMING_ENABLE 1
#define DEFAULT_HOMING_DIR_MASK 17
#define DEFAULT_HOMING_FEED_RATE 200.0 // mm/min
#define DEFAULT_HOMING_SEEK_RATE 2000.0 // mm/min
#define DEFAULT_HOMING_DEBOUNCE_DELAY 250 // msec (0-65k)
#define DEFAULT_HOMING_PULLOFF 3.0 // mm

#define DEFAULT_SPINDLE_RPM_MAX 1000.0 // rpm
#define DEFAULT_SPINDLE_RPM_MIN 0.0 // rpm

#define DEFAULT_LASER_MODE 0 // false

#define DEFAULT_X_STEPS_PER_MM 400.0
#define DEFAULT_Y_STEPS_PER_MM 400.0
#define DEFAULT_Z_STEPS_PER_MM 100.0 // This is percent in servo mode
#define DEFAULT_A_STEPS_PER_MM 400.0
#define DEFAULT_B_STEPS_PER_MM 400.0
#define DEFAULT_C_STEPS_PER_MM 100.0 // This is percent in servo mode

#define DEFAULT_X_MAX_RATE 30000.0 // mm/min
#define DEFAULT_Y_MAX_RATE 30000.0 // mm/min
#define DEFAULT_Z_MAX_RATE 8000.0 // mm/min
#define DEFAULT_A_MAX_RATE 30000.0 // mm/min
#define DEFAULT_B_MAX_RATE 30000.0 // mm/min
#define DEFAULT_C_MAX_RATE 8000.0 // mm/min

#define DEFAULT_X_ACCELERATION (1500.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
#define DEFAULT_Y_ACCELERATION (1500.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
#define DEFAULT_Z_ACCELERATION (100.0*60*60)
#define DEFAULT_A_ACCELERATION (1500.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
#define DEFAULT_B_ACCELERATION (1500.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
#define DEFAULT_C_ACCELERATION (100.0*60*60)

#define DEFAULT_X_MAX_TRAVEL 250.0 // mm NOTE: Must be a positive value.
#define DEFAULT_Y_MAX_TRAVEL 250.0 // mm NOTE: Must be a positive value.
#define DEFAULT_Z_MAX_TRAVEL 100.0 // This is percent in servo mode 
#define DEFAULT_A_MAX_TRAVEL 250.0 // mm NOTE: Must be a positive value.
#define DEFAULT_B_MAX_TRAVEL 250.0 // mm NOTE: Must be a positive value.
#define DEFAULT_C_MAX_TRAVEL 100.0 // This is percent in servo mode       

#endif
```
I copied and edited pinout for 6-axis machine (CPU_MAP_FOO_6X). Axis X to Z work OK. I need to mention that I setup my board with dual functionality. This mean that I have some I/O pins acting like A-DIR for example and if this axis is not used, I thought that I could use this pin for servo. Is this OK? I mean, if I do not define it in cpu_map.h it should be OK to use I/O pin for something else, right?

bdring, 2020-03-22:
You should be able to swap between stepper motor and servo features on an I/O pin. It is best to comment out both step and direction when using that axis as a servo.  

You might want to use #ifdef's to define the each mode of the dual modes to prevent mistakes when switching.

We did some major changes today. We no longer use cpu_maps and use machine definitions instead. This was done to clean up the code and require less editing by the user. **You should create your own mega_shaped.h file and reference that in machine.h and recompile and upload.**

I would start with a definition like this for the servo

```
#define SERVO_B_PIN       GPIO_NUM_15
#define SERVO_B_CHANNEL_NUM   4
#define SERVO_B_RANGE_MIN   0.0
#define SERVO_B_RANGE_MAX   10.0
#define SERVO_B_HOMING_TYPE SERVO_HOMING_TARGET
#define SERVO_B_HOME_POS    SERVO_B_RANGE_MAX // move to max during homing
#define SERVO_B_MPOS        false   // will not use mpos, uses work coordinates

#define DEFAULT_B_STEPS_PER_MM 100.0 // this is used as a calibration for servos
#define DEFAULT_B_MAX_TRAVEL 100.0 // this is used as a calibration for servos

```

for testing....

Turn homing off with **$22=0**
Disable stepper idle with **$1=255**
Send **$rst=#** to remove any work offsets
Restart grbl_esp32 with **[ESP444]RESTART**

Servo should move to one end of travel.
It should resist manual turning
Send **G0B10 ** It should move to the other end of travel
Send **G0B5** It should move to the middle of travel.

bdring, 2020-03-24:
I made some updates to the firmware and wiki to help with this.  

Closing Issue

closed 2020-03-24 by bdring