bdring/Grbl_Esp32 #361: Circular includes of grbl.h?

alejandromango, 2020-03-23:
I am working on a fork to implement a new custom machine based on the custom_machine_template. I have found that the inverse kinematics function relies on a circular include of the grbl.h file. Specifically, the plan_line_data_t struct is undefined when the compiler enters the custom_machine_template.h file, but by a nested series of #include "grbl.h" the struct is defined by the time it is used. Adding a header guard to the grbl.h file breaks custom machines that use inverse kinematics (tested with commit a9cc851 using the CPU_MAP_POLAR_COASTER config setting).
 
Unfortunately, NOT having a header guard breaks my custom machine code. My instinct is to just add a header guard to grbl.h, but I don't have nearly deep enough knowledge of the code to know what other effects that might have. Was leaving out the header guard a conscious decision, and is there a good documentation somewhere of the funky dependencies it creates?

Sorry for the long post, I hope it is clear. Thanks so much for this great software, I'm super excited to  get it fully ported to the machine I'm working with.

alejandromango, 2020-03-23:
Wow, just updated my fork and it looks like there have been a ton of changes to custom machines in the last 30 days. I'll have to test more to see if I am still encountering this error

MitchBradley, 2020-03-23:
I was the culprit behind the refactoring of cpu_map into the Machines/ framework, so I'll be interested to hear how you fare.  I'll help with any issues that you encounter.

closed 2020-03-28 by bdring