bdring/Grbl_Esp32 #362: CNC Shield?

nsiatras, 2020-03-23:
Hello,

Kind of newbie in the ESP32 platform..
I want to ask if this firmware works with the Arduino Nano CNC Shield v4 ?
Is it compatible ?

Thank you

MitchBradley, 2020-03-23:
It only works on the ESP32 processor.  There are some compatible shields on bdring's tindie store. Anything with Arduino Nano would have to use the regular Grbl, not Grbl_ESP32.

nsiatras, 2020-03-23:
I know it works with ESP32 processor. My quesion is ... if I put the ESP32 dev board on the Arduino Nano CNC shield will it work ? Or I have to change the pinout through the code?

MitchBradley, 2020-03-23:
I'm not sure how you would put the ESP32 board onto a shield for a Nano.  ESP32 dev boards are longer, wider, have more pins, run at 3.3V instead of 5V, etc.  You would have to make an adapter board that handles all those problems.  If you did make such a board, you could route the pins however you wanted.  The effort of making the adapter board would be similar to the effort of making a complete shield for ESP32.

bdring, 2020-03-23:
The closest thing to Arduino compatibility is the ESPDuino-32. You could use that with some of the Arduino UNO CNC Shields.

[image]

bdring, 2020-03-23:
There was some discussion of a [Nano adapter board here](https://github.com/bdring/Grbl_Esp32/issues/12#issuecomment-414566409).

[image]

misan, 2020-03-23:
But you won't be able to program it with the CNC Shield on.

ithinkido, 2020-03-24:
> The closest thing to Arduino compatibility is the ESPDuino-32. You could use that with some of the Arduino UNO CNC Shields.
> 
> [image]

I have used this combo before for a 2 axis + servo Z axis.Works great with the V3 shield. I needed to remove the pull up resistor for the stepper enable pin, but otherwise no problems.
[image]

This would be the machine definition for that shield
https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/Machines/espduino.h

derFrickler, 2020-03-24:
Yes, the ESPDuino works fine, but lacks the SD-Card. 
Its allthough easy to build one on your own using an Arduino Proto Shield:
https://der-frickler.net/_media/technik/millycnc/milly_cnc_electronics_002.jpg
Solder an ESP32 Board and an SD-Card Adapter and wire everything to the Connectors on the outside.

misan, 2020-03-24:
This ESP32-UNO board does include the SD card

https://www.uctronics.com/wiki/index.php/(SKU:_B0105)Arducam_ESP32_UNO_Board_for_Arducam_Mini_Camera_Module_Compatible_with_Arduino_UNO_R3

nsiatras, 2020-03-24:
So if I plug a cnc shield on an ESPDuino it will work?
Is the pinout ready ?

nsiatras, 2020-03-24:
Well.. I just opened the "Machines" folder and I saw a pinout for the ESPDuino so... I just ordered an ESPDuino to test it. Thank you for your help guys!

nsiatras, 2020-03-27:
Well.. I bought a WEMOS D1 R2 board.
Is this compatible ?

nsiatras, 2020-03-27:
I am trying to build the sketch and upload to the WEMOS D1 R2 Board.
But I get the following error:
**A fatal error occurred: Timed out waiting for packet header**

For the board build I selected the "WEMOS D1 MINI ESP32" because I cannot find anything closer to the D1 R2 board on the list.

Any ideas?

bdring, 2020-03-27:
Either the serial port is wrong or this is the [old boot button issue](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware#other-firmware-upload-errors). I don't think that board has a boot button though.

Try changing the board type to the first one in the list.  "ESP32 Dev module"

What does the supplier say to do?

nsiatras, 2020-03-27:
If I install the boards from here:
*http://arduino.esp8266.com/stable/package_esp8266com_index.json
<http://arduino.esp8266.com/stable/package_esp8266com_index.json>*

Then I am able to upload a blink sketch to the board so it is not the old
boot button issue

*Let me check with the ESP32 Dev Module*

bdring, 2020-03-27:
If that is an ESP8266 controller, Grbl_ESP32 will not run on it.

nsiatras, 2020-03-27:
I see...
So I need an other board

bdring, 2020-03-27:
I can sell you one pre-programmed and ready to run.

https://www.tindie.com/products/33366583/grbl_esp32-cnc-development-board-v41/

nsiatras, 2020-03-27:
Nice board but I need one for a different setup that is why I am looking for one simular to Arduino uno.

I think this one might work
https://www.aliexpress.com/item/32991841801.html

closed 2020-04-04 by bdring