bdring/Grbl_Esp32 #365: Sending G-code commands from custom_code_template.cpp

kolins-cz, 2020-03-26:
Hi,
I am developing antenna tracker for FPV with stepper motors. Until now I have controlled it by Marlin firmware on old RAMPS. But yesterday I discovered Grbl_Esp32. I have made board for it and now it works. It works great! I would like put actual antenna tracker code inside grbl_esp32. 

**Please describe the feature you would like implemented**
I looked trough custom_code_template.cpp and other custom codes but I have not found example how to send Gcode commands from the inside. It will be really great if you can provide more advanced example for custom code. 

**Why do you think this would improve Grbl_ESP32?**
A little. 

**What do you need the feature for?**
To control motion from the inside. ESP32 should have enough resources to do other tasks

**Will this feature appear to a lot of users?**
Maybe.

MitchBradley, 2020-03-26:
How about using the macro facility in the WebUI to send your custom code?

kolins-cz, 2020-03-26:
Macro won't do. I need to calculate azimuth/inclination in real time and update position every say 200ms.
It may be something else than gcode, I can just call some function.  But code is very complex and I have hard time to identify which one is the right one.

I have noticed I also need some function which will be called periodically from main code.  Also I need access to one serial port.

I know that my idea is kind of wierd. I can always use second microcontroller for tracking stuff and just send Gcode via serial in normal way. But I think ESP32 is powerful enough to calculate some angles and do the usual grbl stuff at same time. 

If you say it is generally bad idea I will just place second microcontroller and use Grbl as is.
Thanks Mirek

bdring, 2020-03-26:
Look at the [polar_coaster.cpp file](https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/Custom/polar_coaster.cpp#L229).

It shows several features you could use.

MitchBradley, 2020-03-26:
The GCode motion commands all boil down to calls to mc_line() and mc_arc().

MitchBradley, 2020-03-26:
For executing something periodically, you could add a subroutine call to the loop in protocol.cpp: protocol_main_loop().

Here is some information about using additional serial ports:
https://quadmeup.com/arduino-esp32-and-3-hardware-serial-ports/

bdring, 2020-03-26:
The atari plotter project is another good example of how your machine can control itself through gcode. It implements tool changing and a custom homing sequence.

https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/Custom/atari_1020.cpp

kolins-cz, 2020-03-27:
Thank you for your input.

closed 2020-03-27 by kolins-cz