bdring/Grbl_Esp32 #367: Crash after long move of rotational axis

dawidma81, 2020-03-28:
Hi,
I am not sure if this a real bug or just some unintended usage of GRBL.

I have a router lathe (linear X and Z axis, a rotational A axis, no Y axis). The A axis has no hard limits, it can spin can spin beyond 360 degree.

The machine is configured as a 4 axis machine (X, Y, Z, A), just Y is left unconnected. Hard and soft limits are disabled to get unlimited axis travel for A.

To make the wooden logs round, i use a simple GCODE script that spins the A axis continuously while feeding the X axis. At the beginning and the end of the log, the Z axis gets lowered while keeping A running:

```
F1200
G0 X0 Z0 A0
G1 A200 Z-7 (lower z axis while rotating material)
G1 X330 A14760 (long spiral move forward; 41 full rotations)
G1 Z-14 A15120 (lowering z again with complete rotation)
G1 X0 A0 (long spiral move backwards --> here it fails)
G0 Z0
```
The script runs fine until it reaches the second last line. There the machine crashes:
- The X axis moves towards 0, but much too fast.
- The A axis seems to move backwards
- The Z axis tries to move somehow, but faster than it can physically follow

The problem seems to be repeatable.

Is this a overflow due to the long A travel? If yes, what maximum travel is considered to be safe?
Is there a better way to implement a rotational axis?

Running only the forward spiral (X and A incrementing) does not cause any problems even doing it several times with resetting the A work coordinate to 0 in between the runs, even though this leads to a further increase of the A axis machine coordinates.

Used version: d159dc2 (24.03.2020)

Thanks for your support!

bdring, 2020-03-28:
The position is stored in a signed 32 bit number in steps. It is not likely to overflow. 

Can you post your machine file and default settings.

dawidma81, 2020-03-28:
Thanks for your quick response.
Here are the settings:
```
$$
$0=3
$1=208
$2=0
$3=4
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=3
$24=200.000
$25=500.000
$26=250
$27=5.000
$30=1000.000
$31=0.000
$32=0
$100=480.000
$101=100.000
$102=480.000
$103=142.222
$110=3000.000
$111=1000.000
$112=1000.000
$113=4000.000
$120=80.000
$121=200.000
$122=80.000
$123=200.000
$130=300.000
$131=300.000
$132=300.000
$133=300.000
```

[routerlathe.h.txt](https://github.com/bdring/Grbl_Esp32/files/4397801/routerlathe.h.txt)

dawidma81, 2020-03-29:
I disconnected the electronics from the machine and ran the same GCODE again while watching the output of the Webinterface and the values seemed to be all fine. Also I did not see any strange messages on the serial port of the ESP32.
GRBL output:
```
...
<Run|MPos:337.648,0.000,-132.400,28424.357|FS:1200,0|Pn:XZ>
<Run|MPos:339.008,0.000,-132.400,28484.312|FS:1200,0|Pn:XZ>
<Run|MPos:340.367,0.000,-132.400,28544.318|FS:1200,0|Pn:XZ>
<Run|MPos:341.725,0.000,-132.400,28604.238|FS:1200,0|Pn:XZ>
<Run|MPos:343.085,0.000,-132.400,28664.236|FS:1200,0|Pn:XZ>
<Run|MPos:344.446,0.000,-132.400,28724.234|FS:1200,0|Pn:XZ>
<Run|MPos:345.806,0.000,-132.400,28784.252|FS:1200,0|Pn:XZ>
<Run|MPos:347.165,0.000,-132.400,28844.193|FS:1200,0|Pn:XZ|Ov:100,100,100>
<Run|MPos:348.525,0.000,-132.400,28904.256|FS:1200,0|Pn:XZ>
<Run|MPos:349.883,0.000,-132.400,28964.133|FS:1200,0|Pn:XZ>                  ==> end of forward spiral
<Run|MPos:351.154,0.000,-132.483,29024.539|FS:1200,0|Pn:XZ>                  ==> downward feed
<Run|MPos:351.154,0.000,-133.650,29084.494|FS:1200,0|Pn:XZ|WCO:21.154,0.000,-125.400,14260.241>
<Run|MPos:351.154,0.000,-134.817,29144.576|FS:1200,0|Pn:XZ>
<Run|MPos:351.154,0.000,-135.983,29204.477|FS:1200,0|Pn:XZ>
<Run|MPos:351.154,0.000,-137.150,29264.475|FS:1200,0|Pn:XZ>
<Run|MPos:351.154,0.000,-138.315,29324.436|FS:1200,0|Pn:XZ>
<Run|MPos:351.104,0.000,-139.400,29377.943|FS:1200,0|Pn:XZ>                   ==> backward spiral
<Run|MPos:349.796,0.000,-139.400,29318.023|FS:1200,0|Pn:XZ>
<Run|MPos:348.485,0.000,-139.400,29257.984|FS:1200,0|Pn:XZ>
<Run|MPos:347.175,0.000,-139.400,29197.943|FS:1200,0|Pn:XZ>
<Run|MPos:345.856,0.000,-139.400,29137.461|FS:1200,0|Pn:XZ>
<Run|MPos:344.544,0.000,-139.400,29077.316|FS:1200,0|Pn:XZ>
<Run|MPos:343.233,0.000,-139.400,29017.318|FS:1200,0|Pn:XZ>
<Run|MPos:341.923,0.000,-139.400,28957.293|FS:1200,0|Pn:XZ>
<Run|MPos:340.612,0.000,-139.400,28897.266|FS:1200,0|Pn:XZ>
<Run|MPos:339.300,0.000,-139.400,28837.072|FS:1200,0|Pn:XZ>
<Run|MPos:337.992,0.000,-139.400,28777.131|FS:1200,0|Pn:XZ>
```

After connecting the electronics back again, I did a dry run of the machine. There it crashed again at the same point (begin of backward spiral).
I made a short video showing the end of the forward spiral, the downfeed and the crash:
https://drive.google.com/file/d/1OrfLY5QPACYSnLTPtPbGddDO2J6nypRV/view?usp=sharing

Do you have any ideas?

bdring, 2020-03-29:
By "crash" do you mean the ESP32 resets? Can you capture the error report on the serial port?

dawidma81, 2020-03-29:
The ESP32 does not reset also looking at the output of the webconsole everything seems fine, but the axis get actuated incorrecty.
The sound you hear in the video starting at 9s is coming mainly from the Z axis which is shaking (faster actuted than the motors can follow), even though there is no Z movement command at all. Also the X axis is moving faster backwards than it should.

dawidma81, 2020-03-31:
The problem can also be reproduced with small simultanuous movements of the A and X axis:

```
ESP Reset
Home
F1000
G1 X10 A10   --> ok
G1 X0        --> ok
G1 A0        --> ok

ESP Reset
Home
F1000
G1 X10 A10   --> ok
G1 X0 A0     --> Z axis tries to move uncontrolled/rattles
```
So the length of the move does not seem to matter.
Jogging all axis independent of each other does work without problems.

Is the problem may be caused by the choice of my pinning?

dawidma81, 2020-04-01:
Hi Bart,
I did some measurements with the oscilloscope and realized that the problem is not caused by GRBL or the ESP32.
Since I use HY-DIV268N-5A stepper motor drivers, I need 5V levels for EN, STEP and DIR. To convert the 3.3V levels of the ESP to 5V logic, I used TXB0108 (or chinese clones) level shifters.
Depending on the movements of X and A, the driver for the Z axis seems to pick up pulses of the other channels. This caused the strange behaviour.
I replaced the level shifters with a 8 channel darlington array (ULN2803A) and now it works like a charm.
Sorry for bothering you with that problem. Keep up your work. Your doing a fantastic job with this project!

closed 2020-04-01 by dawidma81