bdring/Grbl_Esp32 #368: Z Probe Pin Error

jaysettle, 2020-03-30:
Sorry if this was mentioned before, I've done some googling but still having a problem.

I get:
>>> G38.2 F50 Z-5
[Error] An error was detected while sending 'G38.2 F50 Z-5': (ALARM:4) Probe fail. Probe is not in the expected initial state before starting probe cycle when G38.2 and G38.3 is not triggered and G38.4 and G38.5 is triggered. Streaming has been paused.

when trying to probe with the Z Axis.

My probe pin is on GPIO 35 as the config.h states.
[image]

What am I missing. I'm assuming that pin will need to go to ground when the Z Axis touches?

Thanks.

bdring, 2020-03-30:
On a serial monitor send the "?" (status) command with the probe touching and not touching. Paste the results in a reply.

jaysettle, 2020-03-31:
Here you are.. Did I do this correct? Thanks again.

[image]

derFrickler, 2020-03-31:
You probe pin is always on, thats the problem. 
I guess its due to the fact that GPIO35 has no internal pull-up.
you need to add a pullup to 3.3V to turn it "off" when not touching.
see: https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins#input-only-no-pulluppulldown

bdring, 2020-03-31:
for some reason your pin is not changing state when going from the untouched state to the touched state. Check the voltage on GPIO35.

jaysettle, 2020-03-31:
First 3 seconds of the video it's touching and voltage is 0.
After 3 seconds it's not touching and voltage is in the 100mV range.

I think this supports derFrickler's comment. I could either change the pin in the firmware to maybe GPIO 33 and move the actual probe wiring to that pin, or add a pull-up resistor. Does this sound correct?

Please let me know if you can't view the video because of google drive permissions.
https://photos.app.goo.gl/roevKhYX3mE8rbcH6

MitchBradley, 2020-03-31:
Either way - adding the pullup or moving to gpio 33 - should work.  Just try it.

bdring, 2020-03-31:
Can you send a picture of your controller setup?

jaysettle, 2020-04-01:
Yes the pullup resistor got it working thank you.

No judgement on my controller it's working!
[image]

[image]

bdring, 2020-04-01:
Great

closed 2020-04-01 by bdring