bdring/Grbl_Esp32 #377: 4 TMC5160 drivers SPI daisy chain with auto squaring and sensorless homing?

bennedebakker, 2020-04-12:
Hi,

First of all, thank you for the amazing work you guys have been doing for this project! I am in the process of designing a simple PCB for 4 TMC5160 drivers in daisy chain mode. (Something very similar to https://github.com/bdring/4_Axis_SPI_CNC).

I would like it to have the following functions:

- 1 stepper motor on the X-AXIS, 2 stepper motors on the Y-AXIS and 1 stepper motor on the Z-AXIS
- sensorless homing (if possible)
- auto squaring of the gantry, hence two stepper motors on the Y-AXIS
- configuration of the drivers through SPI via daisy chain

So far I understand that I will have to use 1 CS pin for all the drivers and a separate STEP pin for the second Y-AXIS motor. To get auto squaring to work, I would need to connect two limit switches (1 for each Y-AXIS motor), to the same Y_LIMIT_PIN, correct? machine.h  would then look something like this:

`#define MACHINE_NAME "TEST_CNC"

#define USE_TRINAMIC
#define TRINAMIC_DAISY_CHAIN

#define USE_TRINAMIC_ENABLE

#define USE_GANGED_AXES // allow two motors on an axis

#define X_DRIVER_TMC5160
#define X_RSENSE                0.075f
#define X_STEP_PIN      		GPIO_NUM_12
#define X_DIRECTION_PIN 		GPIO_NUM_26
#define X_TRINAMIC
#define X_CS_PIN				GPIO_NUM_17

#define Y_DRIVER_TMC5160
#define Y_RSENSE                0.075f
#define Y_STEP_PIN      		GPIO_NUM_14
#define Y2_STEP_PIN     		GPIO_NUM_21  // ganged motor
#define Y_DIRECTION_PIN 		GPIO_NUM_25
#define Y_TRINAMIC
#define Y_CS_PIN				X_CS_PIN
#define Y_AXIS_SQUARING

#define Z_DRIVER_TMC5160
#define Z_RSENSE                0.075f
#define Z_STEP_PIN      		GPIO_NUM_27
#define Z_DIRECTION_PIN 		GPIO_NUM_33
#define Z_TRINAMIC
#define Z_CS_PIN				X_CS_PIN

#define X_LIMIT_PIN             GPIO_NUM_2
#define Y_LIMIT_PIN             GPIO_NUM_4
#define Z_LIMIT_PIN             GPIO_NUM_15`

So the second Y-AXIS motor is defined by Y2_STEP_PIN. For the SPI setup, it looks like you need to put the drivers in the right order, i.e. X, Y then Z when using daisy chain mode. How do you set up a second Y-AXIS motor with the same settings as the first one, where does it fit in the daisy chain? Do you have to wire it after the first Y, so X-Y-Y2-Z, or at the end, as if it was the A-AXIS X-Y-Z-Y2? Is there automatically an extra set of bits send when you enable USE_GANGED_AXIS and Y2_STEP_PIN, so (4) sets of 40 bits?

One last question: Is sensorless homing already supported by GRBL_Esp32, i.e. connect diag1 or diag0 to the limit switch pins?

Thank you for any help.

Benne

bdring, 2020-04-12:
FWIW: I have a large batch of those boards coming soon. Already shipped via DHL, but they are slow these days.

Ganged axes with Trinamic drivers is not implemented yet. I will add that this week. The order of the drivers is currently specified like this. I'll need to make this dynamic to allow ganged axes to be inserted. Most likely a typical order will be like X-Y-Y2-Z.

```C++
#ifndef TRINAMIC_DAISY_CHAIN
    #define X_DRIVER_SPI_INDEX -1
    #define Y_DRIVER_SPI_INDEX -1
    #define Z_DRIVER_SPI_INDEX -1
    #define A_DRIVER_SPI_INDEX -1
    #define B_DRIVER_SPI_INDEX -1
    #define C_DRIVER_SPI_INDEX -1
#else
    #define X_DRIVER_SPI_INDEX 1
    #define Y_DRIVER_SPI_INDEX 2
    #define Z_DRIVER_SPI_INDEX 3
    #define A_DRIVER_SPI_INDEX 4
    #define B_DRIVER_SPI_INDEX 5
    #define C_DRIVER_SPI_INDEX 6
```

My board is designed for 4 completely independent axes, so it would also need a Y2_DIRECTION_PIN. That gives the controller more flexibility, but you would not need to do that on your controller.

Generic Sensorless homing is not supported in the master branch, but has been demonstrated on other branched like the [wallbot](https://github.com/bdring/Grbl_Esp32/blob/Wallbot/Grbl_Esp32/wallbot.cpp).

Right now the basic features of the Trinamic motors are setup by #defines. This is not very scalable and makes a mess of the code when trying to implement too many features. This is a global problem inherited from AVR Grbl that we are working on.

There is also the issue of tuning the motors. The wallbot code can do this. On my controller, I have some jumpers that can select between switches and the diag pins on the drivers. I would suggest doing that and experimenting with the sensorless homing.

Note: In my experience sensorless homing is accurate to about 1 full step at best. In most cases a switch will be more accurate.

bennedebakker, 2020-04-12:
Oh awesome, thank you for the quick reply Bart! I will definitely check it out on Tindie once it's available. It looks like the board will support everything I need when ganged axes with trinamic drivers is added, so probably no need to design my own. 

So the $$ settings for the X or Y axis would then simply be copied to the second driver when ganged axes is enabled right? I don't think there is ever a need for separate settings for the two drivers of 1 axis.

bdring, 2020-04-12:
There can only be one set of values for resolution, acceleration, etc per lettered axis.

bdring, 2020-04-12:
I made the changes and pushed the code to the devt branch. That will be merged to the master branch soon.

It all worked fine. I am glad this came up so I could knock it out before the boards arrived.

closed 2020-04-27 by bdring