bdring/Grbl_Esp32 #385: Problems with control switches reset, hold and start.

Antonio-Albano, 2020-04-26:
Hello all,

I assembled one board based on Bart Dring's project ESP32 CNC Controller V3.1.
I tested so far the spindle, coolant and all motors and every thing works well.
Now I am trying to use the control switches reset, hold and start buttons don't work, I measured the GPIO 34, 36, 39 voltage when the button is open and it is 3.291V and when I press the button the voltage goes to 33.3mV. I believe that this voltage when the button is pressed is low enough to  the ESP32 understand as low. 

The configuration in CPUmap.h is:
#define CONTROL_RESET_PIN         GPIO_NUM_34  // needs external pullup
#define CONTROL_FEED_HOLD_PIN     GPIO_NUM_36  // needs external pullup
#define CONTROL_CYCLE_START_PIN   GPIO_NUM_39  // needs external pullup

I am using three normal open switches, and Universal Gcode Sender or Web UI to control the machine.
I have no idea what is happening, I read the wiki page but I didn't find anything that could help me. 
Can someone help me with this problem please?

[image]

Best regards,

Antonio Albano.

bdring, 2020-04-26:
Make sure 
``` C++
#define #define IGNORE_CONTROL_PINS
```

is not on somewhere. Search config.h and your machine definition file.

Antonio-Albano, 2020-04-26:
Thanks a lot for the fast answer, everything is working, all three buttons and emergency stop.
Now I just need test the limits switches, but I believe that it will work without problems.
Regards,
Antonio Albano.

closed 2020-04-26 by Antonio-Albano