bdring/Grbl_Esp32 #387: Problems Compiling Firmware

BlueOrangeLive, 2020-04-28:
Please answer the following questions:

Have you read the [wiki regarding how to compile](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)? Yes

What version of the Arduino IDE are you using? 1.8.13

What version (commit date) of the [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) are you using?  1.0.4

Are you using the master branch of Grbl_ESP32? Yes

Have you made any edits or configuration changes (list them) to the  firmware? No
Download 28.04.2020
I have problems with the current version when compiling today. 
The previous version can be compiled.

Please paste the compiler error text here:

In file included from sketch/grbl.h:56:0,
                 from sketch/serial.h:24,
                 from sketch/grbl.h:55,
                 from sketch/report.h:24,
                 from sketch/grbl.h:54,
                 from sketch/motion_control.h:30,
                 from sketch/grbl.h:50,
                 from sketch/grbl_eeprom.h:24,
                 from sketch/grbl.h:47,
                 from sketch/system.h:23,
                 from sketch/grbl.h:43,
                 from sketch/settings.h:28,
                 from sketch/grbl.h:42,
                 from sketch/tdef.h:4,
                 from sketch/grbl.h:39,
                 from /home/juergn/Arduino/Grbl_Esp32-master/Grbl_Esp32/Grbl_Esp32.ino:21:
sketch/tools/SpindleClass.h:41:18: fatal error: grbl.h: No such file or directory
compilation terminated.
exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.

bdring, 2020-04-28:
I can confirm this is a problem.

This is an Arduino IDE issue. I do all of my development in VSCode with platformio. I tested the compile in Arduino IDE, but somehow it was not using the latest code and I did not notice. It did it again just now.

It has to do with code in sub folders. I'll try to do a quick fix and clean it up later.

bdring, 2020-04-28:
@BlueOrangeLive 

It should be working now.

closed 2020-04-28 by bdring