bdring/Grbl_Esp32 #391: ENABLE_SAFETY_DOOR_INPUT_PIN

yellobello, 2020-05-04:
Hi,

Is it OK meanwhile to undefine this? 
(a comment in config.h says "Leave this enabled for now .. code for undefined not ready"
I do not have such a door and find it a little bit annoying.
It does compile fine when undefined and apparently grbl works, but I do not want any strange unexpected behaviour to pop up out of the blue.

In case I build a case with a door, can the corresponding pin be directly switched to GND without causing the µC pin to fry or do I need to use a pulldown resistor?

bdring, 2020-05-04:
All of [these pins](https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins#input-only-no-pulluppulldown) are input only and have no pull up. 

If you define an input on them and they have no pullup, they could randomly and rapidly fire. This could cause a lot of problems and crashes.

It is OK to undefine any I/O pin assignment.

Note: There is currently a problem with the spindle code in the master branch that will cause problems and crashes. We are trying to fix it or roll back the changes.

bdring, 2020-05-04:
BTW: the master branch has been reverted to a working version.

closed 2020-05-04 by bdring