bdring/Grbl_Esp32 #392: 2 noob questions: axis deceleration different from acceleration - running custom G-code?

iacopo, 2020-05-04:
Hello,
I'm a complete noob in many aspects, this is my first project on an esp32, first time I use a stepper motor and the list goes on, so please forgive me if I'm asking stupid things. A google search didn't help out so far.

To put things in context: I have a video production company and my goal is to make a set of versatile robot-like tools to be used in videomaking, the first few seconds of this video should give you a rough idea https://www.youtube.com/watch?v=kzEU3wBVPXs

My questions are the following:
I've set up a custom machine.h file and I can run some stepper motors with no problem. I'm manually building gcode files and I'm able to vary the acceleration rate even during the gcode program by entering a $120 value between the lines, that is fantastic, but would there be a way to implement a different deceleration rate?
Basically I'd like to be able to set an axis at $120 value and decelerate at $xyz value, is that possible at the moment or would it require a lot of custom coding? If so, could you point me in the right direction? I'm learning a lot from this project, and I'm not afraid by its challanges

Second question: is it possible to create custom G code commands and then tell grbl to pass that command to a certain custom function I will define in a custom code section following the custo_code_template.cpp
My aim is to be able to control RGB lights (array of 5050 LEDs) on or off, change color, change intensity value etc... I've previously made arduino controllers for this kind of lights, but I'm lost here as I have no idea how to tap into the G code side of things.

I hope this is the right place to ask such questions :)

Thanks!

bdring, 2020-05-04:
You cannot control the acceleration on the fly from gcode. Any time the machine is idle, you can change the $120 settings.

Grbl_ESP32 only supports standard gcodes. You could hack in your own, but we do not want non standard gcode. The [LinuxCNC project](http://linuxcnc.org/docs/html/gcode/overview.html) is a good source for standard gcode. We are not interested in supporting RepRap gcode.

iacopo, 2020-05-04:
Thanks for your reply, I see the point in avoiding non standard gcode, I gave a look at the linuxCNC project already and I see theres a set of user defined M-code: M100-M199, how would one program a function to be executed when such a command is entered?
I guess that would be pretty common on CNC machines if someone adds a custom tool for instance.

bdring, 2020-05-04:
You would have to add them to the firmware. It is not very common to add arrays of lights to CNC machines.

Grbl_ESP32 has a few commands that branch off to user control. For example tool changing. Grbl_ESP32 does not know how to control your tool changer, but if a user supplied a function to do that, Grbl_ESP32 will optionally pass the tool change information to your function and wait until the function completes.

I suppose that could be done with M100-M199. with that said it is probably a 5-6 hour project for someone very familiar with Grbl to do it right. New modal groups would need to be added, etc. 

I can add that to the roadmap, but I don't think any of the primary developers are going to see that as a priority.

iacopo, 2020-05-04:
I wasn't expecting you or other devs to do this for me, I will try to tackle it, I’ve spent a couple of hours reading the code and it’s very well commented so I’ll give it a try :)

misan, 2020-05-05:
@iacopo you might want to have a look at Marlin firmware. It has a few of the things you are asking:
1) you can change acceleration values on the fly (only to affect to future moves), but by design, the same value is used for acceleration and deceleration https://marlinfw.org/docs/gcode/M201.html
however, if you split a single movement into a sequence of small moves, each one at a different speed, you can create the illusion of a given acceleration (or deceleration), as motion planning does not need to stop in between moves.
2) you can set values to output pins using g-code https://marlinfw.org/docs/gcode/M042.html 
3) you can control RGB lights (WS8212 strips) from g-code too https://marlinfw.org/docs/gcode/M150.html

But, by the sound of it, custom code might be what you need so you have all the flexibility you want, some input to synch with the outer world, and then code that just gives one or more steppers the pulses with the right timing.

iacopo, 2020-05-05:
@misan Thank you very much for your reply, it seems like you pointed me in the right direction with Marlin firmware.
I'll try customizing Marlin to my needs.
I gave up with custom coding from scratch because I found out grbl drives stepper motors way more smoothly than the conventional stepper libraries found for arduino.

terjeio, 2020-05-05:
@iacopo  If you want to use 32-bit processors, such as ESP-32, then my port/rewrite of grbl has the option to use plugin code to extend the M-code range supported - here is a [template](https://github.com/terjeio/grblHAL/blob/master/templates/mcodes.c) that can be used as a starting point. Parsing, modal group handling and synchronization (if required) is taken care of in the core so the code required can be quite simple.

Also be aware that changing settings with $-commands will trigger a flash erase/rewrite of the settings area. My port buffers the settings area and by default delays writing until grbl is in a idle state which may help reducing flash wear. Adding a M-code to switch writing to flash on/off is also easy to implement as the write call is passed over a function pointer.

closed 2020-11-02 by bdring