bdring/Grbl_Esp32 #394: IntelliSense with VSC and PlattformIO

floggy22, 2020-05-06:
A stupid question but i have no idea what im doing wrong. IntelliSense do not work with Grbl_Esp32. I have a new installation with VSC 1.44.2, PlattformIo (core 4.3.3, home 3.2.1), Microsoft C/C++ extension 0.28.0-insiders2 and nothing else. And the framework for esp32 of course. Compiling and building works perfect. With "simple" projects intellisense works. Maybe someone has an advice for me.

bdring, 2020-05-06:
Recently VS Code is a mess. It has flagged over 50 errors, including things like missing semicolons, that are there and compile fine.

[image]

What a mess, but code is fine.

floggy22, 2020-05-06:
OK, than i am not stupid at all. And what do you use for development?

bdring, 2020-05-06:
VS Code. It is faster than Arduino. 

I hate the broken code folding too.

bdring, 2020-05-07:
@MitchBradley cleaned it up.

Thanks!

commit 129f14b4 (2020-05-07): Suppress Intellisense problem reports Fixes #394

The .cpp files for spindles are included into
one large compilation unit that shares a .h
file, so it should not be necessary to
#include SpindleClass.h in each one, but
Intellisense doesn't know that so it issues
numerous problem reports. The solution
is to #include SpindleClass.h in each
Spindle*.cpp file, letting the header guard
prevent multiple inclusion during compilation.
commit 2b15b586 (2020-05-07): Merge pull request #396 from MitchBradley/Devt

Suppress Intellisense problem reports Fixes #394
closed 2020-05-07 by bdring