bdring/Grbl_Esp32 #403: bCNC transmitter error reboot

BlueOrangeLive, 2020-05-09:
Hello bdring
With the new version from yesterday the bCNC transmitter is no longer running (bCNC-0.9.14.306).
The ESP is rebooted endlessly.
mfg Jürgen

[bCNC_ESP32_Problem.txt](https://github.com/bdring/Grbl_Esp32/files/4604244/bCNC_ESP32_Problem.txt)

floggy22, 2020-05-09:
It only crashes when wifi is enabled.

floggy22, 2020-05-09:
The Backtrace Log:

Backtrace: 0x400def59:0x3ffb3e40 0x400df1d1:0x3ffb3fa0 0x400df392:0x3ffb3fc0 0x4008a165:0x3ffb3fe0

Decoding stack results
0x400def59: report_realtime_status(unsigned char) at C:\Users\flogg\AppData\Local\Temp\arduino_build_159193\sketch\report.cpp line 731
0x400df1d1: execute_realtime_command(unsigned char, unsigned char) at C:\Users\flogg\AppData\Local\Temp\arduino_build_159193\sketch\serial.cpp line 208
0x400df392: serialCheckTask(void*) at C:\Users\flogg\AppData\Local\Temp\arduino_build_159193\sketch\serial.cpp line 134
0x4008a165: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

bdring, 2020-05-09:
Can you post your machine definition file?

BlueOrangeLive, 2020-05-09:
Here is my machin.h file
[machine.h.txt](https://github.com/bdring/Grbl_Esp32/files/4604306/machine.h.txt)

BlueOrangeLive, 2020-05-09:
Here is my machine file hjh_coco_cnc.h file
[hjh_coco_cnc.h.txt](https://github.com/bdring/Grbl_Esp32/files/4604312/hjh_coco_cnc.h.txt)

bdring, 2020-05-09:
I think I found the issue. It was a timing issue, that I did not see during testing. The wifi starts before the spindle which sometimes slowed the creation of the spindle object. If a sender is already connected to the serial port, it could ask the spindle for status. This is a call to a null, which causes a crash like you were seeing.

Try compiling the Devt version of the firmware as see if the problem is fixed.

BlueOrangeLive, 2020-05-09:
I just copied the ino file.
Everything works again with bCNC and in the browser.
Tomorrow I will test it more intensively because we have 10:45 p.m.
My milling machine is on the first floor of an apartment building.
Thank you bdring for the quick solution.
mfg Jürgen

BlueOrangeLive, 2020-05-10:
Hello bdring
For me, the connection with bCNC and WiFi is OK.
I thank you again for the quick solution.
Greetings from Kevelaer in Germany and have a nice day
  Jürgen.

closed 2020-05-10 by BlueOrangeLive