bdring/Grbl_Esp32 #405: ESP-Restart after pressing hold and then stop during a job

BlueOrangeLive, 2020-05-11:
Hello dbring
Pausing and then stopping the job while it is running leads to a connection break in the WiFi. The ESP32 then reboots, notice it then that it changes from ap to st mode.
Use Machine Test Drive or my own.
Use today's code.
mfg Jürgen

bdring, 2020-05-11:
What program and method are you using to pause and stop the running job?

BlueOrangeLive, 2020-05-11:
I used the ESP3D Web UI in the browser.
Your test files on SD card and started from there.
Then at the top of the menu Grbl in the line with RUN "II" and then the red button. Then the connection error occurs.
In the No Laser Mode the file tree_laser_mode.nc then shows Error 20 after the start, that's OK, but the process bar remains at the top. The red button under must then still be pressed.

bdring, 2020-05-11:
I cannot repeat the error.

Can you open a serial port. Click the reset button to capture the boot information, then run the things you do to cause the error. The serial port should log what is going on.

Can you get on the slack channel?

https://join.slack.com/t/buildlog/shared_invite/zt-7twbithi-bRGScOaCgJoBpTbkykfxoA

BlueOrangeLive, 2020-05-11:
[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]

Grbl 1.2a ['$' for help]

Here I pressed stop

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400e0a5a  PS      : 0x00060a30  A0      : 0x800de278  A1      : 0x3ffd2e30  
A2      : 0x3ffc6aa8  A3      : 0x3ffc6af4  A4      : 0x00000000  A5      : 0x3ffc5e90  
A6      : 0x3ffc5a10  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x3ffd2e10  
A10     : 0x3f000000  A11     : 0x00000018  A12     : 0x3ffc6494  A13     : 0x3ffba750  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400e0a5a:0x3ffd2e30 0x400de275:0x3ffd2ea0 0x400de2a7:0x3ffd2ec0 0x400de867:0x3ffd2ef0 0x400e1917:0x3ffd2f10 0x400dc44f:0x3ffd2f30 0x400dfdc5:0x3ffd2f50 0x400dffe2:0x3ffd2f70 0x4008ef2d:0x3ffd2f90

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[MSG:Grbl_ESP32 Ver 1.2a Date 20200507]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:Test Drive - Demo Only No I/O!]
[MSG:Axis count 3]
[MSG:Timed Steps]
[MSG:No spindle]

[MSG:Client Started]
[MSG:Connecting Negrui2018]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connected with 192.168.178.48]
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]

Grbl 1.2a ['$' for help]

I can understand English well, but I cannot write. Google writes for me.

closed 2020-05-18 by BlueOrangeLive